greiginsydney/Intervalometerator #28: Enclosure design

AnatolySoloviov, 2019-08-15:
Hi. I can't message Greig or anyone else so I'm leaving it here.
For me it would be interesting to work on enclosure design of such device. I know, it's low-cost DIY thing but we can consider 3D printing.
My idea is to make sealed enclosure with mounting features and decent structural strength. I don't know if anyone wants it. Just an idea.

greiginsydney, 2019-08-18:
Hey @AnatolySoloviov, we'd welcome that for sure. I guess part of the challenge is to make it able to fit a range of cameras and batteries.

Hopefully others will add some comments here to see if there's sufficient interest.

- Greig.

AnatolySoloviov, 2019-08-19:
Sometimes I'm doing design of things that will never be produced. Just for fun, it's different from doing job.

I don't know about different cameras, maybe they have some unified standard and we'll be able to mount them in relatively similar way inside. In worst case we can try to mount everything (camera module and lens) by bayonet. I'd say at first we should consider most optimal and cost-effective hardware set and then check which other cameras we can put there with adjustments.

I will need info about hardware geometry.

AnatolySoloviov, 2019-08-22:
Is there a way to use only necessary parts of the camera? If we put entire camera there, the enclosure needed only to protect Rpi and battery. I understand that for DIY project it is simple to put complete camera in box. Maybe I overestimated the need of new enclosure.
I just wanted to make something interesting, but now I think it may end up as another abandoned design. It wouldn't compete with widely available box with holes.

BTW, if the device uses Rpi, did you consider special camera modules instead of consumer cameras? There's a lot of builds based on something like Sony IMX sensors, MFT or smaller. In some cases it may be more cost effective than using consumer product. I don't know about sensor size requirements for particular application. I'm just thinking about using essential components.

Anyway. If we put camera, Rpi and battery, what exactly we should improve comparing to current box? Sealing for the lens? More flexible and universal external mounting of enclosure?

OzRocky, 2019-08-23:
The core of this project was for it to be able to use a standard DSLR camera for a long term time-lapse, exploiting the quality of the larger sensors and range of lenses. There are already quite a few commercial products out there based around small sensors ranging from just a couple of hundred dollars up. The problem is that the small sensors with their basic lenses just can’t deliver the image quality of a DSLR.

So any box will need to be able to contain a DSLR with its lens, the circuit board and battery. So far, the best option I’ve come up with is to use a Pelican case. In fact, many of the high priced commercial DSLR time-lapse systems are also built in modified Pelican cases. 

The hardest part about using the Pelican case is the opening for the lens. Gluing in a standard lens adapter to screw a filter to is the best option that I could come up with, but there may well be better ways of doing it.

closed 2019-09-08 by greiginsydney