greiginsydney/Intervalometerator #41: 502 Bad Gateway

cadpix, 2020-02-15:
Hi there, 
Really keen to get the Intervalometerator working! What a great project! 
Have built everything onto the pcb and are now having one minor issue. 
After following the Raspberry setup instructions to the dot and then pointing my browser at the pi IP I am greeted with '502 Bad Gateway' nginx/1.14.2.
I have repeated the install process twice to be sure. 
I have restarted nginx and given it time to boot but still have the same issue.
Have also done : sudo nginx -t
Get response:
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful

Directing the browser to ip:8080 shows up 'Welcome to nginx! If you see this page, the nginx web server is successfully installed and working. Further configuration is required.'
Would be grateful if you could assist in getting me past this bump in the setup!
Thanks

greiginsydney, 2020-02-16:
Hey thanks for your interest in the project, and sorry for the bugs. Hopefully this won't be too hard to resolve.

Check out the gunicorn.error file @ /home/pi/www. It will usually point to the offending line number of the code bug. Please post the results here in case it's something that needs to be fixed back in the repo.

- Greig.

cadpix, 2020-02-16:
Hi Greig,

Thanks for the help, I've had a look at the log and I can see a few errors in there so thought might as well share it here as you will probably have a better idea of what to do than me.

Let me know if you need any more details, happy to contribute where I can.

[2020-02-15 17:33:56 +0000] [423] [DEBUG] Current configuration:
  config: None
  bind: ['unix:intvlm8r.sock']
  backlog: 2048
  workers: 3
  worker_class: sync
  threads: 1
  worker_connections: 1000
  max_requests: 0
  max_requests_jitter: 0
  timeout: 300
  graceful_timeout: 30
  keepalive: 2
  limit_request_line: 4094
  limit_request_fields: 100
  limit_request_field_size: 8190
  reload: True
  reload_engine: auto
  reload_extra_files: []
  spew: False
  check_config: False
  preload_app: False
  sendfile: None
  reuse_port: False
  chdir: /home/pi/www
  daemon: False
  raw_env: []
  pidfile: None
  worker_tmp_dir: None
  user: 1000
  group: 33
  umask: 7
  initgroups: False
  tmp_upload_dir: None
  secure_scheme_headers: {'X-FORWARDED-PROTOCOL': 'ssl', 'X-FORWARDED-PROTO': 'https', 'X-FORWA$
  forwarded_allow_ips: ['127.0.0.1']
  accesslog: None
  disable_redirect_access_to_syslog: False
  access_log_format: %(h)s %(l)s %(u)s %(t)s "%(r)s" %(s)s %(b)s "%(f)s" "%(a)s"
  errorlog: /home/pi/www/gunicorn.error
  loglevel: debug
  capture_output: False
  logger_class: gunicorn.glogging.Logger
  logconfig: None
  logconfig_dict: {}
  syslog_addr: udp://localhost:514
  syslog: False
  syslog_prefix: None
  syslog_facility: user
  enable_stdio_inheritance: False
  statsd_host: None
  dogstatsd_tags:
  statsd_prefix:
  proc_name: None
  default_proc_name: wsgi:app
  pythonpath: None
  paste: None
  on_starting: <function OnStarting.on_starting at 0xb5dac4f8>
  on_reload: <function OnReload.on_reload at 0xb5dac588>
  when_ready: <function WhenReady.when_ready at 0xb5dac618>
pre_fork: <function Prefork.pre_fork at 0xb5dac6f0>
  post_fork: <function Postfork.post_fork at 0xb5dac780>
  post_worker_init: <function PostWorkerInit.post_worker_init at 0xb5dac858>
  worker_int: <function WorkerInt.worker_int at 0xb5dac8e8>
[... 820 lines cut ...]
  File "/usr/local/lib/python3.7/dist-packages/gunicorn/workers/base.py", line 144, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/usr/local/lib/python3.7/dist-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/usr/local/lib/python3.7/dist-packages/gunicorn/app/wsgiapp.py", line 49, in load
    return self.load_wsgiapp()
  File "/usr/local/lib/python3.7/dist-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgia$
    return util.import_app(self.app_uri)
  File "/usr/local/lib/python3.7/dist-packages/gunicorn/util.py", line 358, in import_app
    mod = importlib.import_module(module)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/www/wsgi.py", line 17, in <module>
    from intvlm8r import app
  File "/home/pi/www/intvlm8r.py", line 40, in <module>
    from werkzeug.contrib.cache import SimpleCache
ModuleNotFoundError: No module named 'werkzeug.contrib'
[2020-02-15 21:39:45 +0000] [350] [INFO] Worker exiting (pid: 350)
[2020-02-15 21:39:45 +0000] [351] [INFO] Worker exiting (pid: 351)

greiginsydney, 2020-02-17:
Ahhh, the joys of open source.

[Incompatibility issues with werkzeug 1.0](https://github.com/fengsp/flask-session/issues/112)
"On February 6th a new version of Werkzeug was released: https://pypi.org/project/Werkzeug/#history
It introduces a breaking API change which results in a crash."

Until that can be resolved the fix appears to be to downgrade back to Werkzeug 0.16.0 to work around this issue [Reference](https://github.com/Azure-Samples/ms-identity-python-webapp/issues/16#issuecomment-586004077):

```text
sudo pip3 uninstall Werkzeug
sudo pip3 install Werkzeug==0.16.0
```

I'll update the repo now to temporarily force 0.16.0 and keep an eye out for a proper fix.

Thanks for letting me know.

\- G.

cadpix, 2020-02-18:
Hi Greig,

Spot on, works perfectly.

I will continue the build!

Thanks!

greiginsydney, 2020-02-18:
Excellent news. 

I'm monitoring the issue on the Werkzeug/FLASK issues lists and I'll leave this one open here until I can remove the band-aid.

- G.

closed 2020-07-04 by greiginsydney