greiginsydney/Intervalometerator #48: Image transfer from camera to pi intermittant

cadpix, 2020-07-13:
When copying images from the camera to the pi I am getting some strange behaviour. 
Either the transfer simply doesn't happen at the preset time OR it will only copy 84 images when there could be say 100 on there to copy. 
If I use the 'copy now' button it will copy but only 84 images at a time. 
I am using a Canon 6D and this problem has only really become apparent. 
Before I copy the logs etc I thought I would check to see if anyone else had had this happen?

greiginsydney, 2020-07-13:
I've not seen that one before. Eighty-four repeatedly?

I've previously tested and confirmed that the camera will stay awake during the copy process, so it shouldn't be that.

Let me load up a CF with images and see if I can dupe it. If you get a chance, please have a peek at the gunicorn.error and cameraTransfer.log files for anything obvious and paste them in here?

I'm stuck on another project at the moment but should be able to refocus onto this before the week's out. 

\- G.

OzRocky, 2020-07-14:
> Either the transfer simply doesn't happen at the preset time OR it will only copy 84 images when there could be say 100 on there to copy.

Can you let me know what image format you have the camera set to, and I'll try to duplicate the problem with my 6D?

r.

cadpix, 2020-07-14:
Hi both,

The 6d is set to av (aperture priority) and it is shooting raw images only. The camera is set to sleep after 60 seconds.

Everything else looks to be working as it should and the camera wakes when requested. 

I will check out the error and logs and come back to you : )

w

cadpix, 2020-07-14:
Could this be the culprit perhaps? Have found this at the end of each of the 'batches' of copied images in gunicorn.errors

[2020-07-13 10:12:05 +0100] [8764] [DEBUG] Deleted /store_00020001/DCIM/100CANON/_MG_7210.CR2
[2020-07-13 10:12:05 +0100] [8764] [DEBUG] Copying /store_00020001/DCIM/100CANON/_MG_7211.CR2 --> /home/pi/photos/DCIM/100CANON$
[2020-07-13 10:12:05 +0100] [384] [CRITICAL] WORKER TIMEOUT (pid:8764)
[2020-07-13 10:12:07 +0100] [17710] [INFO] Booting worker with pid: 17710
[2020-07-13 10:12:10 +0100] [17710] [DEBUG] hiddenTransferOptions =
[2020-07-13 10:37:06 +0100] [8834] [DEBUG] GET /
[2020-07-13 10:37:06 +0100] [4210] [DEBUG] GET /login
[2020-07-13 10:37:06 +0100] [4210] [DEBUG] Its a GET to LOGIN
[2020-07-13 10:37:06 +0100] [8834] [DEBUG] GET /static/main.css
[2020-07-13 10:37:06 +0100] [8834] [DEBUG] GET /static/favicon.ico
[2020-07-13 10:37:10 +0100] [8834] [DEBUG] POST /login

OzRocky, 2020-07-16:
Yup. I've confirmed the "worker timeout" error when transferring a lot of large images. Hopefully it won't be too difficult a fix.

r.

cadpix, 2020-07-16:
Good news, let me know if you need any more info / tests from here!

greiginsydney, 2020-07-18:
Yuk. This is going to take some time to fix (properly).

We are hitting a 5-minute timeout, and the work-around is to just extend the value.

1.    Edit /etc/systemd/system/intvlm8r.service and change the `--timeout 300` to a 600:

https://github.com/greiginsydney/Intervalometerator/blob/a270516adda83760222281e20eff8fe1ec0f96be/Raspberry%20Pi/intvlm8r.service#L26

2.    Then restart the service (`sudo systemctl restart intvlm8r`) or just wait for the next reboot (assuming it's not running 24x7 of course).

Catch is we seem to hit a DIFFERENT timeout at around the 8-9 minute mark, so there's no benefit to setting a timeout value of longer than 600. You'll see "Ignoring EPIPE" in gunicorn.error when the copy ends.

Leave it with me and I'll see if I can easily extend the band-aid in the short-term, and I'll embark on what's needed to decouple the image transfer process so it'll be free to run for as long as necessary...

\- G.

greiginsydney, 2020-07-18:
BTW the fix to this will also address Issue #22

cadpix, 2020-07-21:
Looks like the --timeout increase is working.
Our daily transfers are back on track which is fantastic.
We are taking 68 pics a day 7 days a week so this is great for now. 
I will monitor this and let you know how we get on! 

W

cadpix, 2020-07-27:
Update for you, while there has been an improvement with the --timeout increase, there still a backlog of images on the camera still, not quite as many as before however. No big deal as just a matter of pressing the 'copy now' button a few times to clear it as I have not had a look at it for a week. 

W

greiginsydney, 2020-07-27:
I'm working on a proper fix for that, don't worry. It might take me a couple more weeks as it's a doozie.

Currently being distracted by a revamp of the Thumbnails page (although it will tie back into this).

cadpix, 2020-07-27:
No worries at all, thanks for your hard work! Look forward to seeing the thumbnails revamp!

greiginsydney, 2020-11-28:
Fixed in release [v4.0.0](https://github.com/greiginsydney/Intervalometerator/releases/tag/v4.0.0) with the addition of Celery & Redis.

closed 2020-11-28 by greiginsydney

greiginsydney, 2021-02-05:
For the benefit of anyone coming along later, any problems with image backlogs on the camera can be resolved by setting more than one upload schedule per day:

https://github.com/greiginsydney/Intervalometerator/blob/master/docs/FAQ.md#can-i-copytransfer-images-more-than-once-per-day

The only caveat is that the Pi will need to be running at the time of this/these extra schedule(s), so that will probably necessitate "Always On" mode, with the resulting extra current drain.