greiginsydney/Intervalometerator #49: Create thumbnails from RAW images

OzRocky, 2020-07-15:
If the camera is shooting RAW, it would be nice to be able to view the extracted thumbnails to confirm that everything is working. Probably only the smallest preview image is required. 

As a bonus, clicking on the thumbnail could _then_ extract the largest preview image from the RAW image (usually full resolution) and display it. Extracting all the large preview images when the files are downloaded from the camera will consume too much disk space. Asking too much? 

This looks promising: https://www.exiv2.org/

cadpix, 2020-07-16:
This would be a fantastic feature and very useful!
It would probably be worth (if it's simple to add in) displaying a few image data fields with the preview such as capture time, date, iso, aperture and shutter speed as this would be helpful info too.

greiginsydney, 2020-11-28:
This is partly addressed in release [v4.0.0](https://github.com/greiginsydney/Intervalometerator/releases/tag/v4.0.0) with the addition of Celery & Redis, and the new Thumbnails page.

Still to come is the download of a high-res JPG version of the image - not the RAW - from the Thumbnails page.

\- G.

greiginsydney, 2021-07-09:
Resolved in [release 4.1.1](https://github.com/greiginsydney/Intervalometerator/releases/tag/v4.1.1)

closed 2021-07-09 by greiginsydney