greiginsydney/Intervalometerator #56: Google oauth process.

peterjbaylis, 2020-09-22:
Greig;

Thanks for the fix to the piTransfer.py script it now runs as expected.
I am now stuck where the output from the script says:

"Go to this link in your browser:"
[image]

I am not sure how to do this as the pi is not running a browser, or at least not one that I can recognise.
So how do I copy the output of the piTransfer.py script to my PC and then copy the auth code back to the pi script?
 This is probably a question driven by lack of familiarity with linux so my apologies if this should be easier than it appears.

Also is raising this kind of question as an issue the most appropriate way to get help  for these kind of support on the fly questions?

Regards
Peter.

peterjbaylis, 2020-09-22:
Hi Greig;

I think I understand now, if I read the instructions for the Dropbox authority process then it makes sense, the Google process can be completed in a similar way just a few more steps for Google than Dropbox.

Cheers Peter.

greiginsydney, 2020-09-22:
Glad you got it sorted Peter.

I'll see if I can re-word the on-screen text to make it clearer.

BTW, asking questions like this is fine via an Issue, as they then become searchable for anyone who follows.

I'll leave this open until I tweak the text.

\- G.

greiginsydney, 2020-10-06:
Fixed in commit https://github.com/greiginsydney/Intervalometerator/commit/15faafd1ef2dcec86108e2d20b290d7050c1d6bc with some more lines of explanatory/supporting text.

closed 2020-10-06 by greiginsydney