greiginsydney/Intervalometerator #58: daylight savings adjustment for Arduino RTC time.

peterjbaylis, 2020-10-10:
some questions that may turn into an enhancement request.
If the pi is configured to use an NTP time server will it update the Arduino time when daylight savings start or stops?

- Do people generally want to use daylight savings?

- Will people simply adjust the "deadon RTC" timer to account for daylight savings, probably a bit inconvenient if the unit is remote.

- is daylight saving something that most people will just ignore?

Cheers Peter.

greiginsydney, 2020-10-10:
Hi Peter,

Yes, I think the intvlm8r should follow "local time".

When I get around to addressing bug #50 it will ensure that **a network-connected Pi** [caveat] will update the Arduino to follow DST, which isn't happening automatically at the moment.

A network-connected Pi syncs its RTC from the internet each time it boots, and regularly if it's set to 'always run'.

An off-grid Pi is more of a challenge. Handling this would require the user to set a timezone and programmatically calculate the in- and out- of DST. Let me ruminate on this one some more. (It's been in the too-hard basket for a while now).

The whole timezone and DST business warrant a mention on the [FAQ](https://github.com/greiginsydney/Intervalometerator/blob/master/docs/FAQ.md) page, which I'll craft at the same time as the issue above is resolved.

Ta for the prompt.

\- Greig.

greiginsydney, 2021-06-05:
Now tracking the sync of time from NTP -> Pi -> Arduino as Issue #84.

I'll leave this issue open until I close 84 & also update the FAQ page.

greiginsydney, 2021-07-09:
Resolved in [release 4.1.1](https://github.com/greiginsydney/Intervalometerator/releases/tag/v4.1.1)

closed 2021-07-09 by greiginsydney