greiginsydney/Intervalometerator #59: web logon credentials

peterjbaylis, 2020-10-19:
### Step 35 for the Setup the pi script.
When you run the "**sudo -E setup.sh web**"  step you will need to provide a login name and password, what characters are allowed for the login name and for the password?
It looks like only lower case and numbers are allowed, could you add in a note as part of the script to provide some guidance for the acceptable characters for the web login credentials?
Peter.

greiginsydney, 2020-10-21:
Hey Peter, thanks for bringing this to my attention.

I see what I've done there. It was an attempt to make the user name case-insensitive, but I was "less than successful".

As a work-around in the short-term, just delete the ".lower()" from the end of what should be around line 216 in intvlm8r.py:
`username = (str(request.form['username'])).lower() #Don't care for case in a username`

I'll fix this properly shortly and also add some guidance to the doco and/or install script too.

\- G.

greiginsydney, 2021-01-27:
Closing issue, now resolved in release v4.0.1

closed 2021-01-27 by greiginsydney