greiginsydney/Intervalometerator #60: Available Shots is wildly inaccurate

OzRocky, 2020-10-31:
The Intervalometerator with a 60D on our test rig is showing 271,693 large jpeg available shots on a 16GB memory card, which seems somewhat optimistic. (Around 2,000 is more likely.)

I ran some tests using gphoto2 and found the following results: 
 
**600D with 32GB card**
Source | RAW | Jpeg
:------------ | ------------- | -------------
Camera Display | 1197 | 4494
GPhoto2 | 12919 | 48431
 
**6D with 32GB card**
Source | RAW | Jpeg
:------------ | ------------- | -------------
Camera Display | 1101 | 4558
GPhoto2 | 11879 | 49111
 
**60D with 32GB card**
Source | RAW | Jpeg
:------------ | ------------- | -------------
Camera Display | 999 | 999
GPhoto2 | 11694 | 42109
 
The 60D is interesting because a) Obviously it only has a three digit counter internally, and b) The number is a fraction of the result in the test rig that is using a card half the size. 
 
With a 20MP camera, I'd expect around 1,000 RAW images and 4,000 Large Jpegs to fit on a 32GB card. 
 
It seems that the issue is more likely with GPhoto than with the Intervalometerator.

greiginsydney, 2020-12-01:
Ha!

Seems [we've been here before](https://github.com/gphoto/gphoto2/issues/188).

I'm guessing we patched this locally and it never made it into a production release?

I'd best get coding.

greiginsydney, 2020-12-02:
Closed having added get/set of capturetarget in /main and /intervalometer pages - borrowing heavily from Jim's examples yet again.

commit f3d851d0 (2020-12-02): get/set capturetarget to resolve availableshots error

Closes #60
closed 2020-12-02 by greiginsydney