greiginsydney/Intervalometerator #61: Use intervalometer without arduino, just for long duration time-lapses

Jones1403, 2020-11-01:
Hello,
Thanks for this amazing application. This is more of a question than an issue, but is it possible to run this only on the Pi without arduino.

I want to use it to make in house long duration time-lapses.
for example a picture every minute for the next 30 days between 9h and 17h to track a restauration project. So continuous power and network is guaranteed.

Is there anything that would need to be changed to have this functionality?

Thanks in advance

Jona

greiginsydney, 2020-11-01:
Hi Jona,

The Arduino is critical, as it's what wakes and fires the camera on the time schedule. The Pi is only used for the "admin" side of things.

If sourcing and programming the Arduino is a problem, we'll be offering some assembled PCBs via eBay soon - although they might not land there until December.

I've not looked into an option where you can forgo the Arduino and RTC and leave the Pi to do the lot, but it's a great question. 

It will be a significant re-write to build all the timing and camera control into the Pi, but leave it with me and I'll look into it. (If it's do-able it certainly won't see the light of day until 2021 though).

\- G.

Jones1403, 2020-11-02:
Wow what a quick response, thanks for that. Sourcing the arduino/pcb is not a problem, but i must have overlooked the fact that the triggering was done using the arduino.

 I can see from the point of view of power consumption that is definitely the best option. Though that is something I’m not really concerned about. 

I started writing something in python using gphoto2. But then thought maybe someone already invented the wheel before me. ;-)

So I can see that will be a big rewrite. Because all I actually need is the intervalometer capability, the camera control and the web transfer.

I think it might be an interesting development for your project, however as I’ll be needing it very soon, it will not be in time for me.
I’ll continue looking.

Thanks again for the quick response.
I’ll definitely follow your project to see what you’ll be developing in the future.

Regards,

Jona

mmunsk, 2023-01-17:
HI Greig,
Tony here, I've been loking for something like this forever and you've MADE IT, congratulations! You're awesome!
I'm trying to put this to work as a winter project and I was wondering if you have these Asembled PCBs available in ebay as you mention.
I would love to get a fully assembled unit from you to try it out.
I would also love to get in contact with you personally to ask a few other questions in case you have the time and are willing to do so, if so this is my email address (plutonio07@hotmail.com) plutonio cero seven at hotmail dot com ( sorry first time using github so not sure if the platform doesn't allow emails) waiting to hear from you soon.

Warm regards.

Thank you so much!

greiginsydney, 2023-01-17:
Hi Tony,

Great to have you on board! See your inbox ... (You might also then choose to delete or edit your comment so your e-mail address isn't public.)

\- Greig.

Plucky1423, 2023-08-16:
Hi @greiginsydney , me again
Did you have any luck with the pi based re build idea jones has asked above?
I'm looking into this as an option while we await the re arrival of the dead on clock component. any advice or even fox holes to avoid is always welcome. 
thanks
JS

greiginsydney, 2023-08-16:
No, sorry. I like it as a concept, but it comes with LOTS of complexities, not the least of which is figuring out how to maintain both operational modes within the same code-base. I'll see if I can give it another review though given the circumstances...

greiginsydney, 2023-08-16:
Actually I'm beginning to think this is a non-starter.

A critical issue is that the Pi doesn't have its own non-volatile real-time clock (RTC), so if it ever loses power, it won't know what the time is when it comes back up. That then necessitates the addition of an external RTC, or a permanent network connection so it can query NTP each time it boots.