greiginsydney/Intervalometerator #67: piTransfer [ftp method] fails to upload files.

peterjbaylis, 2020-11-29:
Hi Greig;

My intervalometerator is not uploading files each day.
The piTransfer log shows that it connects to the ftp server but fails to upload files, 
Here is an extract from the log file. 

**2020/11/29 19:03:16 OK to transfer @ 19:00. Method = FTP
2020/11/29 19:03:16 STATUS: Commencing upload using FTP
2020/11/29 19:03:18 STATUS: No new files to upload**

The "photos/DCIM"  folder on the pi has two directories 100CANON and 101CANON both containing files which have not been uploaded to the ftp server.

The ftp server log file shows that the pi connects each day at ~19:00 but no files get transferred.

Are you able to suggest where I might look to trouble shoot this problem?

Cheers
Peter.

greiginsydney, 2020-11-30:
Hi Peter,

'/home/pi/www/piTransfer.py' uses the function 'list_New_Images' to determine the new images that need to be uploaded.

It looks for all the images in PI_PHOTO_DIR ('/home/pi/photos') and its sub-folders, and discards any that exist in the UPLOADED_PHOTOS_LIST file ('/home/pi/photos/uploadedOK.txt'). Whatever's left it uploads one-by-one, adding each to the UPLOADED_PHOTOS_LIST as it goes.

I can't see anything obvious in the code that might be causing this issue (and none of my recent tests against Google, which uses the same shared 'list_New_Images' function have given me any trouble).

Is there any chance you have swapped the camera's memory card recently, or perhaps your 'File numbering' is set to anything other than 'Continuous'? (We've only ever used Continuous.)

[image]

(The image came from [this support page at Canon](https://support.usa.canon.com/kb/index?page=content&id=ART167999)).

If you've possibly replaced the card with a blank one, the trick is to edit or simply delete the UPLOADED_PHOTOS_LIST file. Beware if you delete it, you'll need to execute these two commands from ssh to re-create it and reset permissions:
```python
touch /home/pi/photos/uploadedOK.txt
sudo chown -R pi:www-data /home/pi/photos/uploadedOK.txt
````
&nbsp; 
\- G.

peterjbaylis, 2020-11-30:
Hey Greig;

Thanks heaps; deleting the **uploadedOK.txt** file and then recreating it has fixed my problem.
I  rebooted the pi after the uploadedOK.txt delete and recreate per your instructions and now the pi is frantically catching up with about 3400 image transfers.
If it runs out of time (60 mins max) then I will just force a reboot and keep doing this until it has caught up.

This issue was made a little tricky to chase down as there is no ftp client on the pi by default. It appears as your intervalometerator code uses an ftp library from python, so this is not available outside the intervalometerator python app.
I was able to test my ftp host by using filezilla in ftp mode so I knew the ftp host (a QNAP NAS box) could accept ftp connections.

Thanks for your quick response on this issue, it has been driving me nuts for over a week, so I'm happy the solution is easy to implement. 

FYI, if any other intervalometerator builders are looking for a robust mount for the Pelican case I found a Vesa mount at Altronics # H 8187 that works nicely for my situation. It is rated for 12Kg and has adjustable friction pivots to allow positioning the case for pan and tilt.

See the image below for my implementation.

[image]

Cheers
Peter

greiginsydney, 2020-12-01:
Nice one Peter! That joint is sufficiently rigid so as not to wobble at all??

BTW, do you realise you don't have to reboot the Pi to copy images?

ssh to the Pi, cd to the www folder and `python3 piTransfer.py copyNow`.

I've not tried it with a VAST amount of images to know if it will run forever or timeout at some stage, so I don't know where your 60 minutes timeout is coming from.

\- Greig.

closed 2020-12-30 by greiginsydney