greiginsydney/Intervalometerator #68: Using feeling

wj7106, 2020-12-23:
After a period of time, we found several problems. First, we can't set the second to shoot. The shortest time is only one minute. Second, there is no way to set multiple time periods in a day, such as shooting from 8:00 a.m. to 12:00 a.m. and then sleeping, shooting from 2:00 p.m. to 9:00 p.m. and then sleeping again. Third, can we add direct transfer button instead of copy in the transmission settings. Fourth, after copying the photos, the USB speed of raspberry PI zero is not fast. After copying raw, the maximum wake-up time of one hour is not enough. It is suggested to change the wake-up time freely. Fifthly, the circuit design should be added to prevent power backflow, and supply power to raspberry pie separately, and the current will return to Arduino. Sixth, add manual photo taking instead of preview function in camera settings, which can make up for missed moments. Seventh, add Baidu network disk transmission function. China can't access Google and other online disks! I am a deep delay photography enthusiast, looking forward to your e-mail communication!

greiginsydney, 2020-12-30:
Hi, thanks for the feedback and sorry it's taken me so long to reply.

1. 60 shots an hour is the maximum we designed the intvlm8r for, and we never thought anyone would take 1 shot per minute. What are you capturing that moves so quickly?
2. We designed the intvlm8r so it would shoot continuously each day ([and hopefully soon overnight](https://github.com/greiginsydney/Intervalometerator/issues/12)). Your application sounds quite specialised if you're wanting to shoot images every 'n' seconds and stop for a 2-hour lunch.
3. What do you mean by "direct transfer"? Do you mean a 1-click way of copying all the images from the camera that then automatically uploads them to the off-Pi destination? 
4. You're right, the Pi zero is not fast, but I wouldn't be surprised if your slow USB feed is a limitation of the camera, not the Pi. If you want to upgrade to a more powerful Pi, we've documented that on the [Advanced Config](https://github.com/greiginsydney/Intervalometerator/blob/master/docs/advancedConfig.md#other-pi-models) page.
I've also just added an [FAQ note](https://github.com/greiginsydney/Intervalometerator/blob/Bugs-2011/docs/FAQ.md#can-i-copytransfer-images-more-than-once-per-day) to show how you can run more than one copy/transfer each day.
5. If you're wanting to power the Pi separately, that's documented at the very bottom of [this page](https://github.com/greiginsydney/Intervalometerator/blob/master/docs/advancedConfig.md#other-pi-models).
6. "Preview" was designed to help you tweak the setup once the box is sealed and in position. How will a manual shoot button "make up for missed moments", especially when you're shooting one shot per minute? Can you explain the use-case?
7. Adding upload to Baidu from here in Australia will be as much as a challenge for me as accessing Google Drive is for you. If you can't find someone locally who can do that for you, the generic FTP and SFTP options should be a good fallback option.

\- Greig.

greiginsydney, 2021-01-27:
Pausing during lunch is now addressed by the "lunchtime kludge" added in 4.0.1. See the [FAQ](https://github.com/greiginsydney/Intervalometerator/blob/master/docs/FAQ.md#can-i-pause-the-shooting-schedule-during-the-day-eg-at-lunchtime) for details.

closed 2021-01-27 by greiginsydney