greiginsydney/Intervalometerator #75: Camer settings doesn't always update to correct settings

aerialvideographyuk, 2021-03-24:
When going to camera settings on the web interface the setting given is the actual setting used for the last picture and not the settings for the camera. If the web browser is refreshed the correct camera settings appear.  This was with a Canon 600D  set to AV, auto ISO and f5.6, auto shutter.  The camera settings displayed were AV, ISO 100, f.5.6, 250 sec.  On refresh the correct settings were shown.

greiginsydney, 2021-03-24:
That's an odd one. I can't say I've seen that here before.

The values are all initialised to blank when the /camera page is called, and then it queries the camera and populates them, so it's not like anything is cached in the intvlm8r.

It's not doing it on the 60D sitting next to me. I can toggle the MF/AF switch and F5 the webpage and the values follow correctly every time.

Does it do this all the time, or is it intermittent?

aerialvideographyuk, 2021-03-24:
It does it all the time but only seems to be when you have to use the "wake the camera" function.
document with screenshots of the sequence of events is attached.
[camera settings web view.docx](https://github.com/greiginsydney/Intervalometerator/files/6197651/camera.settings.web.view.docx)

greiginsydney, 2021-03-25:
Thanks for the doco - I now see i can reproduce this here as well, with the ISO and shutter speed values both changing.

This variation is indeed as a result of waking the camera, because with that half-press it takes a meter reading and determines the correct ISO and shutter speed to use for your given aperture, were you to fully press the button. I note that the camera's display updates from an ISO of "A" to the chosen ISO too.

Let me look into this some more and see what I can do. Thanks for bringing it to our attention.

\- G.

greiginsydney, 2021-08-09:
I've today made some camera-related performance improvements in 4.2.1 that result in faster page load times for /home and /system, and a side bonus is I think it's sufficiently *delayed* the /camera page to suppress this issue.

I'll be monitoring it through this next testing phase. I'll let you know when it's released (ETA 1st Sept) and ask you to let me know if it recurs.

\- G.

greiginsydney, 2021-09-13:
Nah, it's still present. I'm still looking into how best to solve it.

The easy way is to introduce a long delay after the camera is woken and before the camera is queried (waiting for it to timeout) but that's not a bridge I'm ready to cross.