greiginsydney/Intervalometerator #80: Canon's 650d preview ended without response

wj7106, 2021-04-11:
I am now shooting with Canon 500D, Canon 650d and Canon 6D cameras. Canon 650d can't use the preview in the settings. If I use the preview, it will cause no response in 650d. The camera seems to crash when it can't stand by. You must disconnect the USB to use it again. I am now very careful to use the preview function in the camera settings, Canon 500D and Canon 6D will not have this problem. Are the versions of gPhoto2 and libgphoto too low?

greiginsydney, 2021-04-12:
First off I'd check the camera's firmware is the latest. That appears to be [v1.0.5](https://www.canon.com.au/support/sims-content?pid=fd93f047711a47f485b2e68568d5500e&cid=A2A947E320EE48669EC68D3D8AE561EB&ctype=fa). 

As for the intvlm8r, the installation process updates gPhoto2 and libgphoto to their latest, so that's unlikely to be the problem.

What's logged in gunicorn.error when you try the Preview through the intv8lmr's website? Here's mine:
```python
[2021-04-12 10:38:35 +1000] [559] [DEBUG] POST /camera
[2021-04-12 10:38:36 +1000] [559] [DEBUG] -- Camera Preview selected
[2021-04-12 10:38:36 +1000] [559] [DEBUG] Capturing preview image
[2021-04-12 10:38:37 +1000] [559] [DEBUG] [255, 216, 255, 219, 0, 132, 0, 1, 1, 1]
[2021-04-12 10:38:38 +1000] [559] [DEBUG] GET /camera
[2021-04-12 10:38:38 +1000] [556] [DEBUG] GET /static/preview/intvlm8r-preview.jpg
```

To see if the issue is a bug in the intvlm8r's code:
1. ssh to the Pi
2. `cd /usr/local/share/python-gphoto2/examples`
3. wake the camera
4. `python3 preview-image.py`. (You shouldn't need to `sudo` this)

If that works, can you screen-grab any output from the script and paste it back here?

Here's how it looks on my 60D, the first with Raw selected, and then after changing back to 'Large Normal JPEG':

```python
pi@BenchPi3Bplus:/usr/local/share/python-gphoto2/examples $ python3 preview-image.py
Checking camera config
Cannot preview raw images
pi@BenchPi3Bplus:/usr/local/share/python-gphoto2/examples $ python3 preview-image.py
Checking camera config
Capturing preview image
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
<class 'memoryview'> 313372
[255, 216, 255, 219, 0, 132, 0, 1, 1, 1]
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9152 receiving resp failed: PTP Operation Not Supported (0x2005)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x911c receiving resp failed: PTP Device Busy (0x2019)
WARNING: gphoto2: (camera_unprepare_canon_eos_capture [config.c:545]) 'ptp_canon_eos_resetuilock (params)' failed: PTP Device Busy (0x2019)
pi@BenchPi3Bplus:/usr/local/share/python-gphoto2/examples $
```

wj7106, 2021-04-12:
The display code is ：

pi@raspberrypi:/usr/local/share/python-gphoto2/examples $ python3 preview-image.py
Checking camera config
Cannot preview raw images
pi@raspberrypi:/usr/local/share/python-gphoto2/examples $ python3 preview-image.py
Checking camera config
Capturing preview image
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
<class 'memoryview'> 269889
[255, 216, 255, 219, 0, 132, 0, 1, 1, 1]
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9152 receiving resp failed: PTP Operation Not Supported (0x2005)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x911c receiving resp failed: PTP Device Busy (0x2019)
WARNING: gphoto2: (camera_unprepare_canon_eos_capture [config.c:545]) 'ptp_canon_eos_resetuilock (params)' failed: PTP Device Busy (0x2019)

gunicorn:

[2021-04-12 08:59:20 +0800] [413] [DEBUG] POST /camera
[2021-04-12 08:59:22 +0800] [413] [DEBUG] -- Camera Preview selected
[2021-04-12 08:59:22 +0800] [413] [DEBUG] Capturing preview image
[2021-04-12 08:59:22 +0800] [413] [DEBUG] [255, 216, 255, 219, 0, 132, 0, 1, 1, 1]
[2021-04-12 08:59:23 +0800] [414] [DEBUG] GET /camera
[2021-04-12 08:59:26 +0800] [413] [DEBUG] GET /static/preview/intvlm8r-preview.jpg

wj7106, 2021-04-12:
When there is no response, display 
pi@raspberrypi:~ $ python3 preview-image.py
python3: can't open file 'preview-image.py': [Errno 2] No such file or directory
pi@raspberrypi:~ $ python3 preview-image.py
python3: can't open file 'preview-image.py': [Errno 2] No such file or directory
pi@raspberrypi:~ $ python3 preview-image.py
python3: can't open file 'preview-image.py': [Errno 2] No such file or directory
pi@raspberrypi:~ $ python3 preview-image.py
python3: can't open file 'preview-image.py': [Errno 2] No such file or directory
pi@raspberrypi:~ $ python3 preview-image.py
python3: can't open file 'preview-image.py': [Errno 2] No such file or directory
pi@raspberrypi:~ $

greiginsydney, 2021-04-12:
This error:

```python
python3: can't open file 'preview-image.py': [Errno 2] No such file or directory
```

is just indicating you're not in the correct folder.

Try `cd /usr/local/share/python-gphoto2/examples` and go again.

wj7106, 2021-04-12:
pi@raspberrypi:/usr/local/share/python-gphoto2/examples $  python3 preview-image.py
Checking camera config
Capturing preview image
<class 'memoryview'> 326226
[255, 216, 255, 219, 0, 132, 0, 1, 1, 1]
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x9152 receiving resp failed: PTP Operation Not Supported (0x2005)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:514]) PTP_OC 0x911c receiving resp failed: PTP Device Busy (0x2019)
WARNING: gphoto2: (camera_unprepare_canon_eos_capture [config.c:545]) 'ptp_canon_eos_resetuilock (params)' failed: PTP Device Busy (0x2019)

wj7106, 2021-04-13:
I just tested it again. If preview is enabled and a JPG low resolution photo is obtained, but the reflector of the camera is always on and in video mode. Only when you take out the camera battery or unplug the USB cable or set the camera to video shooting and then set the photo mode, the reflector of the camera will turn off and the camera will exit video mode and return to normal, Can automatically 30 seconds standby!

greiginsydney, 2021-04-13:
Is this your problem?

["On Canon EOS / EOS M make sure the preview mode on the camera is in "still image" mode, not in "movie mode"."](http://www.gphoto.org/doc/remote/)

wj7106, 2021-04-13:
Canon 650d uses this system, can't use the preview in the camera settings, if used, the camera's USB has been occupied by the program, the camera can't automatically exit the preview, there seems to be a program in the background has been occupied, can only manually disconnect the USB, or manually take out the battery. Please help me, can only change the camera to solve this problem?

greiginsydney, 2021-04-14:
Given the evidence, your issue is one of bad config in the camera, a bug in the camera's firmware, or a problem in gphoto2 - all of which are outside of my control. (I don't think you've confirmed yet that the camera's firmware is v1.0.5?)

If you can execute the following and send me the output, I'll update the code to disable the Camera page's Preview button on the EOS 650D, and I'll add a note to that effect on the camera compatibilty page.

```python
pi@BenchPi3Bplus:~ $ cd /usr/local/share/python-gphoto2/examples
pi@BenchPi3Bplus:/usr/local/share/python-gphoto2/examples $ python3 camera-summary.py
```

\- G.

wj7106, 2021-04-14:
yes,firmware is V1.0.5

pi@raspberrypi:/usr/local/share/python-gphoto2/examples $ python3 camera-summary.py
Summary
=======
Manufacturer: Canon Inc.
Model: Canon EOS 650D
  Version: 3-1.0.5
  Serial Number: a87c7b4f5e6d41b793b438ad7a8828b
Vendor Extension ID: 0xb (2.0)

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b104), Unknown(b105)

Device Capabilities:
        File Download, File Deletion, File Upload
        No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Capture 2

Storage Devices Summary:
store_00020001:
        StorageDescription: SD
        VolumeLabel: None
        Storage Type: Removable RAM (memory card)
        Filesystemtype: Digital Camera Layout (DCIM)
        Access Capability: Read-Write
        Maximum Capability: 128010158080 (122080 MB)
        Free Space (Bytes): 126523408384 (120662 MB)
        Free Space (Images): -1

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS 650D'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 100% (100)

pi@raspberrypi:/usr/local/share/python-gphoto2/examples $

greiginsydney, 2021-04-14:
OK, the 'blockPreview' code is in. It will be in the next release, to be 4.1.0 (currently the Branch "Bugs-2104").

Thanks for your help with this.

closed 2021-04-14 by greiginsydney

wj7106, 2021-04-14:
I tested for one night. The camera is set to 30 seconds standby. When using the preview function, the camera will not enter the standby mode in 30 seconds. Instead, it adds 5 minutes on the basis of 30 seconds and becomes 5 minutes and 30 seconds standby. As long as it enters the standby mode and wakes up again, the camera will be in normal condition. If the preview function is not used, the camera will automatically enter the standby mode in 30 seconds after taking a picture. This is only a problem Only this 650d will appear, if it is 500D and 6D, using preview, it will enter the standby mode according to the set standby time, without adding 5 minutes!

commit 4b1cb03d (2021-04-14): Added preview blocklist to /camera

Closes #80