greiginsydney/Intervalometerator #82: Time service with Arduino GPS ？

wj7106, 2021-04-19:
Can we use the Arduino GPS timing method to directly control the time when Arduino triggers the camera without raspberry pie board, and use an LCD display with buttons to directly control it, so that we can directly operate and set it in the place without network. After all, we seldom change the parameters after setting the parameters of the camera

greiginsydney, 2021-04-20:
But the Arduino doesn't have a GPS! It doesn't even know what the time is, which is why we had to add the separate real-time clock (RTC) chip/board.

Sure you could connect a GPS 'shield' and do away with the Pi and the RTC, but you'll be losing all of the remote admin capability and the image transfers, and probably also at the expense of the low-power operation we designed in from the outset.

The resulting device wouldn't be an intvl8mr, but simply a different kind of intervalometer.

\- G.

wj7106, 2021-04-20:
OK, I see. Thank you

closed 2021-04-20 by greiginsydney