greiginsydney/Intervalometerator #88: TimeLimitExceeded(1800) Message.

peterjbaylis, 2021-08-05:
Hi Greig;

I am getting a "TimeLimitExceeded(1800)" Message when transferring image files from the camera to the pi.

Here is a screen shot of the message on the web service.

[image]

Can I change this limit to allow the copy and transfer process to complete in one session?

Cheers Peter.

greiginsydney, 2021-08-05:
Hi Peter,

I deliberately baked in a 30-minute timeout so a hung process won't keep holding up resources. It was meant to be a safety-net, but you're not the first person to hit it. Whilst it's easy to change this in intvlm8r.py (roughly around line 1940, depending on the release), it'll be over-written each time you update, so you'd need to remember that:

```python
@celery.task(time_limit=1800, bind=True)
def copyNow(self):
```

My preference is to create one or more extra copy schedules, as outlined in [this FAQ response](https://github.com/greiginsydney/Intervalometerator/blob/master/docs/FAQ.md#can-i-copytransfer-images-more-than-once-per-day), although that's only valid if the Pi is always running.

Let me know which way you go and if that works for you (and doesn't cause other problems).

\- G.

peterjbaylis, 2021-08-07:
Thanks Greig;

I found two instances of this variable at lines 1467 and 1521 in the version of intlvm8r.py that I have installed.
I have changed them both to 7200 to give the pi more time to do its thing.

Thanks for the quick response to this query.

Cheers Pete.

greiginsydney, 2021-08-07:
Without seeing the code to confirm it, I expect the first appearance is the function that copies the files:

```python
@celery.task(time_limit=1800, bind=True)
def copyNow(self):
```

... whilst the second is the bit that creates the thumbnails of them:

```python
@celery.task(time_limit=1800, bind=True)
def newThumbs(self):
```

Let me know how you go.

peterjbaylis, 2021-08-09:
Hi Greig;

Thanks for that info, this has fixed my TimeLimit Exceeded problem.
I have now been able to get all the pi images transferred to my NAS.

**New topic.**
I am getting some forbidden filename messages during the transfer process.
See message below.

_2021/08/08 19:35:39 Uploading /home/pi/photos/DCIM/101CANON/IMG_7952.JPG
2021/08/08 19:35:39 Error uploading 550 IMG_7952.JPG: Forbidden filename,_

I can see these messages in the transfer log.

If I run the piTransfer.py script from the console then it copies some of the files it originally flagged as "Forbidden Filename"
I could get as many as 20-50 of these errors during one transfer session.
If I simply re run the piTransfer script it will successfully copy a few more of the files. 
If I do this multiple times then eventually it copies all of the files from the pi to my NAS.

Not sure if this error is due to a problem with the Raspbian FTP process or it could be my QNAP NAS not playing well with the Raspberry pi.
Have you seen this before using FTP on the pi?

Cheers Pete.

greiginsydney, 2021-08-09:
Hey Pete,

We happen to have a QNAP NAS here and I use that for my FTP testing, and I've encountered no problems with it.

What's your firmware version? Ours is 4.5.4.1741 and we don't update it often. If yours is years behind it might be worth an update.

As the 550 error is coming back from the QNAP and it's intermittent, my first suggestion is to put a pause between each file, in case it's just being overwhelmed. Just add `time.sleep(2)` at the top of the loop, like this (at around line 240):

```python
for needupload in newFiles:
    time.sleep(2)
    log("Uploading " + needupload)
```

If that works out for you I'll add it into the script (and perhaps for all upload types).

\-G.

peterjbaylis, 2021-08-10:
Hi Greig;
 
Adding the sleep time has made a difference it does not fail so frequently now. It is still failing sometimes. I just transferred 120 images and had to restart the transfer process about 6 times for it to complete all the image file transfers.
I will try playing with the sleeptime (x) number to see if extending it to 3 or 4 will eliminate the problem entirely.

Part of my problem is the the pi is very busy as it is trying to build thumbnails for 4000+ images.
Is there some way I can tell the pi to not build the thumbnails or are the thumbnails a dependency for another downstream process?

Pete.

greiginsydney, 2021-08-10:
Hi Pete,

> Part of my problem is the the pi is very busy as it is trying to build thumbnails for 4000+ images.

The thumbnail creation process is only there for the benefit of the /thumbnails page, but copying and thumbs are tasks executed sequentially - they shouldn't be happening in parallel.

If you want to suppress the thumb creation process entirely, just add a 'return' as shown below in intvlm8r.py and 'newThumbs' will abort as soon as it's called:

```python
@celery.task(time_limit=1800, bind=True)
def newThumbs(self):
    return
    app.logger.info('newThumbs() entered') #This logs to /var/log/celery/celery_worker.log
```

Another couple of options for you:
- if you have local power, consider upgrading the Pi from a Zero to a 3 or 4, which will give it a considerable amount of extra grunt. I've documented that [here](https://github.com/greiginsydney/Intervalometerator/blob/master/docs/advancedConfig.md#other-pi-models).
- if the Pi's drowning under old images since transferred, enable 'deleteontransfer' and the Pi will delete them after they've been successfully copied to your FTP server/elsewhere. ([Documented here](https://github.com/greiginsydney/Intervalometerator/blob/master/docs/FAQ.md#my-camera-andor-pi-are-running-low-on-storage-how-can-i-delete-old-images)).

\- G.

closed 2021-08-29 by greiginsydney

peterjbaylis, 2021-08-30:
Greig;

Thanks for the follow up.

Cheers Pete.