greiginsydney/Intervalometerator #89: Preview the problem in raw format

wj7106, 2021-08-07:
You have been shooting in raw format, but if you want to preview, you can only display the photos after setting JPG format, and then set it back to raw format. Can you preview directly in raw format?

greiginsydney, 2021-08-07:
Unfortunately Gphoto doesn't support previewing when you're shooting in raw mode, so your experience is correct.

Try it yourself using one of the example scripts:

```python
pi@BenchPi3BPlus:~/www $ cd /usr/local/share/python-gphoto2/examples
pi@BenchPi3BPlus:/usr/local/share/python-gphoto2/examples $ sudo python3 preview-image.py
Checking camera config
Cannot preview raw images
pi@BenchPi3BPlus:/usr/local/share/python-gphoto2/examples $
```

Leave it with me and I'll see how easy it is to disable the Preview button when you're set for raw, or provide some better user feedback.

\- G.

greiginsydney, 2021-08-08:
It seems this is specific to Canon cameras. The Nikon I have here *does* go through the motions of creating a preview - but all you get on screen is a black image. Safest to disable the button for all...

wj7106, 2021-08-08:
I'm all Canon cameras. This preview function is very good. I can view the shooting situation at any time. I can keep this preview function. If I can't view it directly in raw format, I have to manually switch to JPG format and then switch back to raw!

greiginsydney, 2021-08-08:
Don't worry I'm not removing the feature, just disabling the button when it's not applicable. A popup explains why the button's disabled:

[image]

websocketing, 2021-08-10:
@wj7106 I'm glad to see you pay attention to me. Are you from China? Can you tell me your email, QQ, wechat and other contact information? We can talk.

wj7106, 2021-08-10:
> @wj7106很高兴看到你关注我。你是从中国来的吗？能告诉我你的邮箱、QQ、微信等联系方式吗？我们可以谈。

wj7106@21cn.com

closed 2021-08-11 by wj7106