greiginsydney/Intervalometerator #91: Delete images from camera.

peterjbaylis, 2021-08-28:
Hi Greig;

I have seen the FAQ response which shows how to add  **delete after copy** and **delete after transfer** options.

Is there a command I can run from the pi console to delete camera images directly from the SD card while it is still in the camera?
I have set my interval to one shot every minute so I am getting 600 images per day and even if I only use jpegs then I am having to swap out the camera SD card every 10-12 days.
I will set the delete after copy and delete after transfer options in the **intvlm8r.ini file** but I still want to be able to manually clean out the SD card occasionally.

Cheers Pete.

wj7106, 2021-08-29:
My camera and/or Pi are running low on storage. How can I delete old images?
Hidden config options in the INI file will let you delete images off the camera after they've been safely transferred to the Pi, and after they've been safely uploaded to your nominated "off-box" destination.

This option isn't visible in the browser, so to change it you need to SSH to the Pi. Browse to /home/pi/www/ and edit the file intvlm8r.ini. Set the two highlighted options On or Off as required. If the option's not already there, just add the line in by hand:

[Global]
file created = 05 Jan 2020
thumbscount = 20

[Transfer]
tfrmethod = Dropbox
dbx_token = thequickbrownfoxjumpsoverthelazydogorsomething
transferday = Daily
transferhour = 00
sftpserver = 20.20.20.20
sftpuser = intvlm8r
sftppassword = mysecretsftppassword
sftpremotefolder = upload/GreyBox/
**deleteaftertransfer = On**

[Copy]
copyday = Daily
copyhour = 00
**deleteaftercopy = On**

greiginsydney, 2021-08-29:
Jim's included an example app in python-gphoto2 that will do the job nicely:

```bash
cd /usr/local/share/python-gphoto2/examples
sudo python3 clear-space.py 95
```

The number is telling it the amount of space to free up. Beware I think there's a bug in it: I entered 100 and it loops forever, trying to free up space that's not avalable. So choosing a lower number is a good idea.

greiginsydney, 2021-08-29:
Is this OK to close now Pete?

peterjbaylis, 2021-08-30:
Hi Greig;

The **clear-space.py** script throws the following error;

__pi@picam-03:/usr/local/share/python-gphoto2/examples $ sudo python3 clear-space.py 95
WARNING: gphoto2: (gp_context_error) Could not detect any camera
Traceback (most recent call last):
  File "clear-space.py", line 113, in <module>
    sys.exit(main())
  File "clear-space.py", line 61, in main
    camera.init()
gphoto2.GPhoto2Error: [-105] Unknown model_

I will try waking up the camera first via the intvlm8r web service and have another go.

Cheers Pete.

peterjbaylis, 2021-08-30:
Greig;
Waking up the camera seems to have worked...
The script now runs and deletes images, when the script ends it shows "Camera now has 95.0% free space.

This will save having to extract the card to manually delete images.
I have also set the "**delete aftercopy**" and "**delete aftertransfer**" options in the **intvlm8r.ini** file so house keeping should take a lot less time from this point forward.

Thanks again for your quick turnaround on the question.

Cheers Pete.

peterjbaylis, 2021-08-30:
Greig;

Please close this issue at your convenience.

Cheers Pete.

greiginsydney, 2021-08-30:
Done.

closed 2021-08-30 by greiginsydney