greiginsydney/Intervalometerator #92: About the development board using the same function

wj7106, 2021-08-29:
Because the USB speed of raspberry pi zero is too slow and there is no external wifi antenna, the signal is often bad, so I bought a banana PI bpi-m2+ board at a low price, which has stronger performance than the zero board. See the function introduction, which also matches the GPIO pin function of raspberry pi, but now I have a problem. After the raspberry pi image is installed in banana PI bpi-m2 +, the intvlm8r program is also installed correctly, The GPIO cable is also connected according to the zero pin, but he doesn't work with Arduino and can't communicate with Arduino board. I don't know what went wrong.

greiginsydney, 2021-08-29:
Really hard to know for sure, as it could be any number of incompatibility issues, hardware OR software - or both.

If it can't talk to the Arduino it's probably going ot be related to the I2C bus.

SSH to the Pi and run the setup script with the test option and it might reveal some problems:

```bash
cd ~
sudo ./setup.sh test
```

Make sure the first part of the output matches this:

```bash
TEST!
PASS: /etc/nginx/sites-available/intvlm8r exists
PASS: /etc/systemd/system/intvlm8r.service exists
PASS: /etc/systemd/system/cameraTransfer.service exists
PASS: /etc/systemd/system/piTransfer.service exists
PASS: /etc/systemd/system/setTime.service exists
PASS: /etc/systemd/system/redis.service exists
PASS: /etc/systemd/system/celery.service exists
PASS: i2c_arm_baudrate is present in /boot/config.txt
PASS: i2c-dev installed in /etc/modules

If the Arduino is connected & programmed it will show as "04" in the top line below:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- 04 -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

PASS: The Pi is NOT an AP
```

\- G.

wj7106, 2021-08-29:
PASS: i2c_arm_baudrate is present in /boot/config.txt
PASS: i2c-dev installed in /etc/modules

If the Arduino is connected & programmed it will show as "04" in the top line below:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

PASS: The Pi is NOT an AP

greiginsydney, 2021-08-29:
And the Arduino is connected and awake?

wj7106, 2021-08-29:
Yes, Arduino is powered on and is in emergency mode. Four photos are taken in an hour. The web page can't get the time of the clock module, can't wake up the camera, can't set the interval

greiginsydney, 2021-08-29:
The circuits for both appear to confirm that the I2C interface is on the same pins, so that's a good start.

Are you using a ribbon cable to join the BananaPi to the intvlm8r PCB? Is there any chance that any of the pins are reversed/transposed on the ribbon cable? (I suspect probably not, otherwise you'd have 5V and 3.3V swapped over, which would be quite catastrophic.)

I think it's going to be a challenge to reverse-engineer these sorts of problems with a Pi clone. I'd go Googling for examples of I2C projects for the BananaPi and see if there's any special settings required for it.

wj7106, 2021-08-29:
Looking forward to good news, thank you

wj7106, 2021-08-29:
pi@bpi-iot-ros-ai:~ $ sudo i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- 04 -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@bpi-iot-ros-ai:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@bpi-iot-ros-ai:~ $

wj7106, 2021-08-29:
bus = SMBus(1) #Initalise the I2C bus
# This is the address we setup in the Arduino Program
address = 0x04
 I change SMBus (1) to SMBus (0) and you can use it

greiginsydney, 2021-08-29:
Ha! Looks like they must have accidentally(?) transposed the two I2C buses??

Well done on finding the problem!

closed 2021-08-29 by greiginsydney

wj7106, 2021-08-30:
After testing one night, it was found that he could not sleep according to the set time, nor could he control the restart

greiginsydney, 2021-08-30:
Sorry, not sure I understand you there.

Are you suggesting the Arduino isn't able to shutdown the Pi? If so, that indicates there are more pin incompatibilities in the Banana Pi, or the way I'm using an IO pin to trigger a shutdown isn't supported in its operating system. Check out the `dtoverlay` commands in [setup.sh](https://github.com/greiginsydney/Intervalometerator/blob/master/Raspberry%20Pi/setup.sh).

(I've written separately about that functionality [here](https://greiginsydney.com/pi-shutdown-button/).)

\- G.