greiginsydney/Intervalometerator #101: Hi Greig and Rocky.

wj7106, 2021-11-03:
I want to add a function button to delete camera photos in the transfer page instead of manually SSH to raspberry every time and enter sudo python3 clear-space.py 95 code. How can I add this function? Thank you!

greiginsydney, 2021-11-03:
I've deliberately avoided exposing some of the more "powerful" functionality in the browser interface as a way of reducing the risk of someone doing something they might later regret. This is serves as a security mechanism, limiting the damage that can be done by a malicous actor who gains access to the intvlm8r. 

Check out the [FAQ](https://github.com/greiginsydney/Intervalometerator/blob/master/docs/FAQ.md#my-camera-andor-pi-are-running-low-on-storage-how-can-i-delete-old-images) for how to enable the "DeleteAfterCopy" and "DeleteAfterTransfer" features. These only delete an image AFTER it's been confirmed to have been successfully transferred off the camera or Pi.

Hopefully they'll do what you're after?

\- G.

wj7106, 2021-11-03:
I see. It's very thoughtful. Thank you for your reply!

closed 2021-11-10 by greiginsydney