greiginsydney/Intervalometerator #103: Long-term night shooting

cca8uk, 2021-11-25:
I'm looking around for the best system for my needs, and this is very nearly it, so congrats on a fab project.  

The one additional thing I need for my long-term setup is the ability to shoot only at night.  Setting up a fixed start/end time for triggers therefore isn't practical solution.  To that end I wondered whether you had considered adding the ability to track sunset/sunrise times across the year?  This could be done by either the user having to add a reference table of times/date, or by incorporating a light sensor into the setup/code.

Any thoughts on workarounds or how,/whether this is achievable would be of great interest.

Chris

greiginsydney, 2021-11-25:
Hi Chris,

Thanks for the compliment.

As luck would have it, I'm in final testing of an update I'm calling "STM", aka "shoot through midnight": the ability to set a start time "later" in the day than the end time. (No, it doesn't actually have that at the moment - it's an [issue](https://github.com/greiginsydney/Intervalometerator/issues/12) I've wanted to address for ages). That would give you the crude version of what you're after, but would require tweaking through the seasons, and you'd always end up with some unwanted daylight shots.

Can you tell me more about your application? You want it to start shooting every 'n' minutes from sunset and not stop until sunrise? It sounds like it'd be a real challenge to change the intvlm8r to do that, not the least of which is because the start/stops are all currently based upon an hour value, and not hour:minute - but I'm open to the idea...

\- G.

cca8uk, 2021-11-26:
> CAUTION: This email originated from outside of the organisation. Do not click links or open attachments unless you recognise the sender and know the content is safe.

The STM would be a must for my setup if it’s based on a timer.  I’m basically looking at options for taking photos every ‘n’ minutes between dusk and dawn for the long-term. The photos are of nocturnal insects visiting a light which will be later identified for monitoring purposes.

It’s not against the realms that I could just delete daylight images, but as it’s a remote setup working off a 12v solar/battery, I’m trying to reduce unnecessary power drain everywhere I can.  The nearest hour to sunset/sunrise wouldn’t be too much of an issue, but night length varies from 5 to 18 hours or so here in Scotland across the year, hence the need to somehow track it.

In many ways an incorporated light sensor activating the trigger when a threshold has been crossed could be a simpler solution.  An even simpler method might be to incorporate a light switch onto the incoming power and have the whole thing shut down during the day, but I’m not sure if the Pi reboots itself and the camera (and without corrupting files) if the incoming power is continually interrupted.

Chris

This email and any attachments are intended solely for the named recipients and are confidential. If you are not the intended recipient, please reply to the email to highlight the error and delete this email from your system; you must not use, disclose, copy, or distribute this email or any of its attachments. UK Centre of Ecology & Hydrology (UKCEH) has taken reasonable precautions to minimise risk of this email or any attachments containing viruses or malware, but the recipient should carry out its own virus and malware checks before opening the attachments. UKCEH does not accept any liability for any losses or damages which the recipient may sustain due to presence of any viruses. Opinions, conclusions or other information in this message and attachments that are not related directly to UKCEH business are solely those of the author and do not represent the views of UKCEH. We process your personal data in accordance with our Privacy Notice, available on the UKCEH website. https://www.ceh.ac.uk/privacy-notice Registered office address; Maclean Building Benson Lane, Crowmarsh Gifford, Wallingford, Oxfordshire, United Kingdom, OX10 8BB Companies Registered Name; UK Centre for Ecology & Hydrology Place of Registration; England Registered Company Number; 11314957

greiginsydney, 2021-11-27:
Scotland's on our list of places to get back to. We've only seen a tiny bit of it, no further north than Nairn, and we loved it.

Your suggestion has some merit, but it's not without its challenges.

Benefit:
- allows anyone to shoot from sunrise to sunset or vice versa (or a combo of fixed and calculated)
- the underlying code changes would also let fixed-time users be more granular in their start/stop times: at least to the half-hour.
- if I'm going to this effort, I'd see if I can bake in the lunchtime pause I've been asked about (currently something you hard-code into the Arduino if required).

Cost:
- Change the underlying Arduino/Pi code to send the start/end time as minutes past midnight, rather than just the hour value.
- Requires building in a degree of backwards-compatibility so an upgrade to this code in the Pi doesn't break an Arduino still running an earlier version of code.
- UI changes to allow you to specify lat/long, and the underlying calculation of sunrise and sunset.
- The Pi boot sequence will need to be changed to send "today's" start/stop times, as well as the overnight schedule (for a Pi set to always run). Because the Pi boots only once a day this will probably end up being up to a day out: e.g. it always starts shooting on \*yesterday's\* sunrise, but stops at today's sunset. I'm sure that's an acceptable margin of error. An alternative approach might be to send lat/long to the Arduino and let _it_ do that: more investigation required here (although I'm reluctant to try to ask the poor little Arduino to do more).

If you check out issue #98 you'll see there's a bit of a wish-list queue that I'm working on, so this request might not see the light of day for quite some time. Other than yourself, is there a user-community out there who'll benefit from this? (I'm also looking at adding an external trigger interface for wildlife photographers).

> An even simpler method might be to incorporate a light switch onto the incoming power and have the whole thing shut down during the day, but I’m not sure if the Pi reboots itself and the camera (and without corrupting files) if the incoming power is continually interrupted.

Yeah, you definitely want to avoid yanking the Pi's power without warning.

Let me get STM released, take stock of the dev effort for the next suite of changes, and I'll review this again.

\- G.

greiginsydney, 2021-12-03:
Partially fixed in [release 4.4.0](https://github.com/greiginsydney/Intervalometerator/releases/tag/v4.4.0), which adds STM ("shoot through midnight").