greiginsydney/Intervalometerator #107: Images fail to transfer - due wrap around

aerialvideographyuk, 2021-12-26:
Due to the image names IMG_NNNN the images will have the same names once 9999 images have been exceeded.  New images with the same name as the old ones will not be transferred as the system believes they have already been transferred.  A temporary solution is to delete the file uploadedOK.txt and ensure that the new files are transferred to a different directory.  It would be good if the images had a timestamp added on transfer from camera to pi.  Ie. IMG_YYYYMMDD_HHMM_NNNN or similar.

greiginsydney, 2021-12-28:
Hi Dave,

Thanks for logging this, and well done figuring out the work-around.

I think this issue will potentially manifest twice: once when copying from the camera, and the second time when uploading from the Pi.

The resolution for both is the same: enable 'rename on copy' on the /Transfer page. That will add your preferred prefix or suffix to the image name, ensuring they'll always be unique, even if the camera starts reusing image numbers like you've encountered.

Here's a screen-grab showing how to add a year-month-day-HHMM suffix:

[image]

Can you please try it and let me know if you encounter any problems?
<br>&nbsp;<br>

\- G.

greiginsydney, 2022-02-10:
Closing stale issue

closed 2022-02-10 by greiginsydney