greiginsydney/Intervalometerator #108: lost password after upgrade to 4.4.4

peterjbaylis, 2022-01-01:
Hi Greig;
After upgrading my intervalometerator setup to. 4.4.4 the web password seems to have been clobbered.
I still have console access via ssh.
Where can I setup a new password for the intervalometerator service?

Cheers Peter.

peterjbaylis, 2022-01-01:
Hi Greig;

I found the password stuff lurking in the intvlm8r.py file at line 100
So can I simply edit the intvlm8r.py file to change the web logon credentials?

BTW - Happy New year to you and Rocky.

Cheers Peter.

greiginsydney, 2022-01-01:
Hi Peter,

In 4.2.0 I changed the default password from "password" to a random 8 character password generated when the setup script runs, the point of which being to improve the default security posture. 

If perchance you upgraded but overlooked the `cp ~/www/intvlm8r.py ~/www/intvlm8r.old` step, the code would have treated this as a new install and replaced the default password, as seems to have happened. What's odd though is that if it did that, it should have still prompted you to note or change the password. Any chance you screen-scraped the install/upgrade process (as the script drops a few indicators to screen)?

I've just done a quick test here and it's working as expected - but the 'cp' process is the key to it determining if this is a new install or upgrade..

To change the web login and/or password from ssh, do a `sudo -E ./setup.sh login` and follow your nose.

<br>Ta for the best wishes. I'm thinking a 'happy' new year is perhaps aiming a bit high these days. I'll settle for one that's better than last year.

\- G.