greiginsydney/Intervalometerator #109: cameraTransfer script not executing.

peterjbaylis, 2022-01-10:
Hi Greig,

I have upgraded my intervalometerator to 4.4.0 hoping that it might reduce ftp errors when copying to the NAS.
However since the upgrade it is no longer running the cameraTransfer script so all the images are still sitting on the SD card in the camera.
I have tried clicking the copynow button on the transfer page and also manually running the cameraTransfer script via an ssh session.
Neither of these actions has made any difference.
I also deleted and re added the uploadedOK.txt, although I think that is just for the piTransfer process.

Included here are screenshots from the system maintenance page and the home page of the web interface on the intervalometerator.
[image]

[image]

Can you suggest where I might look to see what the issue withy the cameraTransfer process is?

Cheers Pete.

greiginsydney, 2022-01-11:
Hi Pete,

Sorry it's doing that to you.

'Celery' now does all the heavy lifting, so if it's dead several things won't work.

For a quick test, just run `sudo /home/pi/setup.sh test` and see what comes back.

I'm expecting something like the below about halfway through the output:
```
PASS: nginx    service is running
PASS: intvlm8r service is running
PASS: celery   service is running
PASS: redis    service is running
PASS: remoteit service is not installed
```

If Celery is running OK, we'll need to dig further into the logs. Have a look at the file `/var/log/celery/celery_worker.log`. Search for the text `copyNow entered` (with a datestamp since you've upgraded) and see what's logged immediately after that.

If Celery is dead, it could be pointing to a code bug or some other issue. Try `sudo systemctl status celery` to diagnose Celery problems. You want it to be showing "active (running)" in green like here:

[image]

(Press Q to get out of that display back to the command line).

`journalctl | grep celery` may also shed light on issues with Celery.

Hopefully the above will give us some clues.

\- Greig.

peterjbaylis, 2022-01-11:
Hi Greig;
Looks like celery is not running.

---------------------------------------------------------------------------------------------------------------------------------------
See the outputs from the **setup.sh test** run and **journalctl | grep celery** and **sudo systemctl** status celery included below.

pi@picam-03:~ $ sudo /home/pi/setup.sh test
TEST!
PASS: /etc/nginx/sites-available/intvlm8r exists
PASS: /etc/systemd/system/intvlm8r.service exists
PASS: /etc/systemd/system/cameraTransfer.service exists
PASS: /etc/systemd/system/piTransfer.service exists
PASS: /etc/systemd/system/setTime.service exists
PASS: /etc/systemd/system/redis.service exists
PASS: /etc/systemd/system/celery.service exists
PASS: i2c_arm_baudrate is present in /boot/config.txt
PASS: i2c-dev installed in /etc/modules

If the Arduino is connected & programmed it will show as "04" in the top line below:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- 04 -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

PASS: The Pi is NOT an AP
PASS: WiFi power save is OFF

PASS: nginx    service is running
PASS: intvlm8r service is running
FAIL: celery   service is dead
PASS: redis    service is running
PASS: remoteit service is not installed

Service = setTime
  LoadState   = loaded
  ActiveState = inactive
  SubState    = dead

Service = cameraTransfer
  LoadState   = loaded
  ActiveState = inactive
  SubState    = dead

Service = piTransfer
  LoadState   = loaded
  ActiveState = inactive
  SubState    = dead

Service = heartbeat
  LoadState   = loaded
  ActiveState = active
  SubState    = waiting

NTP is currently active. NTP is our master time source.
------------------------------------------------------------------------------------------------------------------
And here is the output from j**ournalctl | grep celery** and **sudo systemctl status celery** commands.

pi@picam-03:~ $ journalctl | grep celery
Jan 11 19:17:05 picam-03 systemd-tmpfiles[114]: [/etc/tmpfiles.d/celery.conf:1] Line references path below legacy directory /var/run/, updating /var/run/celery → /run/celery; please u                                    pdate the tmpfiles.d/ drop-in file accordingly.
Jan 11 19:17:16 picam-03 systemd-tmpfiles[184]: [/etc/tmpfiles.d/celery.conf:1] Line references path below legacy directory /var/run/, updating /var/run/celery → /run/celery; please u                                    pdate the tmpfiles.d/ drop-in file accordingly.
Jan 11 19:56:03 picam-03 systemd[1]: Starting celery.service...
Jan 11 20:05:05 picam-03 systemd[1]: celery.service: Start operation timed out. Terminating.
Jan 11 20:05:05 picam-03 systemd[1]: celery.service: Control process exited, code=killed, status=15/TERM
Jan 11 20:05:05 picam-03 systemd[1]: celery.service: Failed with result 'timeout'.
Jan 11 20:05:05 picam-03 systemd[1]: Failed to start celery.service.
Jan 11 20:18:02 picam-03 systemd-tmpfiles[811]: [/etc/tmpfiles.d/celery.conf:1] Line references path below legacy directory /var/run/, updating /var/run/celery → /run/celery; please u                                    pdate the tmpfiles.d/ drop-in file accordingly.

------------------------------------------------------------------------------------------------------------------------------------------
pi@picam-03:~ $ sudo systemctl status celery
● celery.service
   Loaded: loaded (/etc/systemd/system/celery.service; enabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Tue 2022-01-11 20:05:05 ACDT; 1h 31min ago
  Process: 399 ExecStart=/bin/sh -c ${CELERY_BIN} -A ${CELERY_APP} multi start --logfile=${CELERYD_LOG_FILE}   --loglevel=${CELERYD_LOG_LEVEL} ${CELERYD_NODES} ${CELERYD_OPTS} (code=killed, signal=TERM)

Jan 11 19:56:03 picam-03 systemd[1]: Starting celery.service...
Jan 11 20:05:05 picam-03 systemd[1]: celery.service: Start operation timed out. Terminating.
Jan 11 20:05:05 picam-03 systemd[1]: celery.service: Control process exited, code=killed, status=15/TERM
Jan 11 20:05:05 picam-03 systemd[1]: celery.service: Failed with result 'timeout'.
Jan 11 20:05:05 picam-03 systemd[1]: Failed to start celery.service.
pi@picam-03:~ $

Hope this sheds some light on where to look!

Cheers Pete.

greiginsydney, 2022-01-11:
Thanks Pete.

I see some improved monitoring and trapping of Celery's health in my future...

"Timeout" doesn't really tell me a lot (and we've been here before with a similar issue with Remote.it, which I see you're not running).

Was there anything in `/var/log/celery/celery_worker.log`?
<br>
\- G.

peterjbaylis, 2022-01-12:
Hi Greig;
There are 2 celery_worker.log files in that directory.
Here is listing of the files. 
They both have the same data stamp. 
The celery_worker.log is 0 bytes long.
The celery_worker.log-20220111 is 34M bytes long.

pi@picam-03:/var/log/celery $ ls -al
total 33820
drwxr-xr-x 2 pi   www-data     4096 Jan 12 19:54 .
drwxr-xr-x 8 root root         4096 Jan 12 00:00 ..
-rw-r--r-- 1 pi   www-data        0 Jan 11 00:00 celery_worker.log
-rw-r--r-- 1 pi   www-data 34622470 Jan 11 00:00 celery_worker.log-20220111

I am assuming you will not want to wade through a 34Meg log file, so which bit will help, I can see some stuff about local host connect refused is that where the log file might provide some clues?

Here is a snip of some of the stuff near to the most recent log entries;
----------------------------------------------------------------------------------------------------------------------------------------
[2022-01-01 14:56:24,073: WARNING/MainProcess] /usr/local/lib/python3.7/dist-packages/celery/worker/consumer/consumer.py:361: CPendingDeprecationWarning: 
In Celery 5.1 we introduced an optional breaking change which
on connection loss cancels all currently executed tasks with late acknowledgement enabled.
These tasks cannot be acknowledged as the connection is gone, and the tasks are automatically redelivered back to the queue.
You can enable this behavior using the worker_cancel_long_running_tasks_on_connection_loss setting.
In Celery 5.1 it is set to False by default. The setting will be set to True by default in Celery 6.0.

  warnings.warn(CANCEL_TASKS_BY_DEFAULT, CPendingDeprecationWarning)

[2022-01-01 14:56:24,474: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/0: Error 111 connecting to localhost:6379. Connection refused..
Trying again in 2.00 seconds... (1/100)

[2022-01-01 14:56:26,599: INFO/MainProcess] Connected to redis://localhost:6379/0
[2022-01-01 14:56:27,035: INFO/MainProcess] mingle: searching for neighbors
[2022-01-01 14:56:28,299: INFO/MainProcess] mingle: all alone
[2022-01-01 14:58:33,624: WARNING/MainProcess] consumer: Connection to broker lost. Trying to re-establish the connection...
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/celery/worker/consumer/consumer.py", line 326, in start
    blueprint.start(self)
  File "/usr/local/lib/python3.7/dist-packages/celery/bootsteps.py", line 116, in start
    step.start(parent)
  File "/usr/local/lib/python3.7/dist-packages/celery/worker/consumer/consumer.py", line 618, in start
    c.loop(*c.loop_args())
  File "/usr/local/lib/python3.7/dist-packages/celery/worker/loops.py", line 81, in asynloop
    next(loop)
  File "/usr/local/lib/python3.7/dist-packages/kombu/asynchronous/hub.py", line 361, in create_loop
    cb(*cbargs)
  File "/usr/local/lib/python3.7/dist-packages/kombu/transport/redis.py", line 1128, in on_readable
    self.cycle.on_readable(fileno)
  File "/usr/local/lib/python3.7/dist-packages/kombu/transport/redis.py", line 398, in on_readable
    chan.handlers[type]()
  File "/usr/local/lib/python3.7/dist-packages/kombu/transport/redis.py", line 733, in _receive
    ret.append(self._receive_one(c))
  File "/usr/local/lib/python3.7/dist-packages/kombu/transport/redis.py", line 743, in _receive_one
    response = c.parse_response()
  File "/usr/local/lib/python3.7/dist-packages/redis/client.py", line 3505, in parse_response
    response = self._execute(conn, conn.read_response)
  File "/usr/local/lib/python3.7/dist-packages/redis/client.py", line 3479, in _execute
    return command(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/redis/connection.py", line 739, in read_response
    response = self._parser.read_response()
  File "/usr/local/lib/python3.7/dist-packages/redis/connection.py", line 324, in read_response
    raw = self._buffer.readline()
  File "/usr/local/lib/python3.7/dist-packages/redis/connection.py", line 256, in readline
    self._read_from_socket()
  File "/usr/local/lib/python3.7/dist-packages/redis/connection.py", line 201, in _read_from_socket
    raise ConnectionError(SERVER_CLOSED_CONNECTION_ERROR)
redis.exceptions.ConnectionError: Connection closed by server.
[2022-01-01 14:58:33,786: WARNING/MainProcess] /usr/local/lib/python3.7/dist-packages/celery/worker/consumer/consumer.py:361: CPendingDeprecationWarning: 
In Celery 5.1 we introduced an optional breaking change which
on connection loss cancels all currently executed tasks with late acknowledgement enabled.
These tasks cannot be acknowledged as the connection is gone, and the tasks are automatically redelivered back to the queue.
[... 2 lines cut ...]

  warnings.warn(CANCEL_TASKS_BY_DEFAULT, CPendingDeprecationWarning)

[2022-01-01 14:58:35,365: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/0: Error 111 connecting to localhost:6379. Connection refused..
Trying again in 2.00 seconds... (1/100)

[2022-01-01 14:58:37,461: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/0: Error 111 connecting to localhost:6379. Connection refused..
Trying again in 4.00 seconds... (2/100)

[2022-01-01 14:58:41,654: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/0: Error 111 connecting to localhost:6379. Connection refused..
Trying again in 6.00 seconds... (3/100)

[... cut ...]
[2022-01-01 14:58:55,777: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/0: Error 111 connecting to localhost:6379. Connection refused..
Trying again in 10.00 seconds... (5/100)

[2022-01-01 14:59:05,849: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/0: Error 111 connecting to localhost:6379. Connection refused..
Trying again in 12.00 seconds... (6/100)

-------------------------------------------------------------------------------------------------------------------------------------------
I can send the complete file via dropbox or wetransfer if you want to see the whole thing.

Cheers Pete.

greiginsydney, 2022-01-14:
Ta Pete, we might need to do that. Let's see if I can avoid the pain of trying to sift through 34M of logs though.

Can you run `celery --version` for me please and just confirm we're on the same version? Here's mine:

```
pi@greyBox:~ $ celery --version
5.0.2 (singularity)
pi@greyBox:~ $
```
<hr/>

Have you installed any other (custom) software components on the Pi? We found with remote.it that we needed to tell Celery to hold off launching until remote.it was up. We could have a similar interaction with yours.

Consider the following a diagnostic step only, not a fix.

Let's see how long systemd is waiting for celery to start:
```
pi@greyBox:~ $ sudo systemctl show celery | grep ^Timeout
TimeoutStartUSec=1min 30s
TimeoutStopUSec=1min 30s
pi@greyBox:~ $
```
Ok, so 90 seconds here. Let's double that to 180.

Edit your celery.service file (`sudo nano /etc/systemd/system/celery.service)` and add a TimeoutStartUSec line to it:
```
[Service]
Type=forking
User=pi
Group=www-data
EnvironmentFile=/etc/default/celery
WorkingDirectory=/home/pi/www
TimeoutStartSec=180

ExecStart=/bin/sh - <etc>
```

Reload the daemon after the change:
```
pi@greyBox:~ $ sudo systemctl daemon-reload
```
Confirm the change:
```
pi@greyBox:~ $ sudo systemctl show celery | grep ^Timeout
TimeoutStartUSec=3min
TimeoutStopUSec=1min 30s
```

Let's see if it works:
```
pi@greyBox:~ $ sudo systemctl restart celery
pi@greyBox:~ $ sudo systemctl status celery
```

So if this starts OK we know it's just a timing problem and there's nothing inherently broken. Mind you, if it doesn't start that *doesn't* definitely point to broken code. (I just upgraded one of my demo setups here to 4.4.0 and Celery is working fine).

\- G.

([Reference](https://sleeplessbeastie.eu/2020/02/29/how-to-prevent-systemd-service-start-operation-from-timing-out/))

peterjbaylis, 2022-01-14:
Hi Greig;

Well done... adding Timeouts to the celery.service file has kicked the cameraTransfer process into action.
I have not forced a reboot of the system yet as I will let it copy the ~ 5000 image files from the SD card first.

Here are some notes I made while running the diagnostics you suggested.
***
[celery diagnostics 20220114.txt](https://github.com/greiginsydney/Intervalometerator/files/7868929/celery.diagnostics.20220114.txt) 

This seems to be where my problem is or was, so does this point directly to where a fix needs to be applied?

Cheers Pete.

peterjbaylis, 2022-01-14:
Hi Greig;

After a reboot the celery service is again reporting as dead. 

***
pi@picam-03:~/photos/DCIM/105CANON $ sudo /home/pi/setup.sh test
TEST!
PASS: /etc/nginx/sites-available/intvlm8r exists
PASS: /etc/systemd/system/intvlm8r.service exists
PASS: /etc/systemd/system/cameraTransfer.service exists
PASS: /etc/systemd/system/piTransfer.service exists
PASS: /etc/systemd/system/setTime.service exists
PASS: /etc/systemd/system/redis.service exists
PASS: /etc/systemd/system/celery.service exists
PASS: i2c_arm_baudrate is present in /boot/config.txt
PASS: i2c-dev installed in /etc/modules

If the Arduino is connected & programmed it will show as "04" in the top line below:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- 04 -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

PASS: The Pi is NOT an AP
PASS: WiFi power save is OFF

PASS: nginx    service is running
PASS: intvlm8r service is running
FAIL: celery   service is dead
PASS: redis    service is running
PASS: remoteit service is not installed

Service = setTime
  LoadState   = loaded
  ActiveState = inactive
  SubState    = dead

Service = cameraTransfer
  LoadState   = loaded
  ActiveState = inactive
  SubState    = dead

Service = piTransfer
  LoadState   = loaded
  ActiveState = activating
  SubState    = start

Service = heartbeat
  LoadState   = loaded
  ActiveState = active
  SubState    = waiting

NTP is currently active. NTP is our master time source.

***
Where else should I be looking or running diags?

Cheers Pete.

peterjbaylis, 2022-01-14:
Hi Greig;

After the reboot;

***
pi@picam-03:~ $ sudo nano /etc/systemd/system/celery.service
pi@picam-03:~ $ sudo systemctl show celery | grep ^Timeout
TimeoutStartUSec=1min 30s
TimeoutStopUSec=1min 30s

***
the celery.service file seems to have gone back to the 90 sec Timeouts. 

I checked with nano and the celery.service file I edited is showing for TimeoutStartUSec=3min
And for TimeoutStopUsec=1min 30s. 

Not sure if this helps on not....

Cheers Pete.

greiginsydney, 2022-01-15:
Well, I think that's a positive outcome to start with.

Try these two changes for me, then reload the daemon and reboot:
- Change the name from 'TimeoutStartUSec' to 'TimeoutStartSec' (ditching the "U")
- Change the value back to raw numerals, e.g. TimeoutStartSec=180

Separate to that, I'm curious to see what other systemd services you might have running. Here's mine (and the command to display them):

```
pi@greyBox:~ $ systemctl --type=service --state=active
UNIT                                                        LOAD   ACTIVE SUB     DESCRIPTION
avahi-daemon.service                                        loaded active running Avahi mDNS/DNS-SD Stack
console-setup.service                                       loaded active exited  Set console font and keymap
cron.service                                                loaded active running Regular background program processing daemon
dbus.service                                                loaded active running D-Bus System Message Bus
dhcpcd.service                                              loaded active running dhcpcd on all interfaces
dphys-swapfile.service                                      loaded active exited  dphys-swapfile - set up, mount/unmount, and delet
fake-hwclock.service                                        loaded active exited  Restore / save the current clock
getty@tty1.service                                          loaded active running Getty on tty1
ifupdown-pre.service                                        loaded active exited  Helper to synchronize boot up for ifupdown
intvlm8r.service                                            loaded active running intvlm8r.service. Gunicorn instance to serve intv
keyboard-setup.service                                      loaded active exited  Set the console keyboard layout
kmod-static-nodes.service                                   loaded active exited  Create list of required static device nodes for t
networking.service                                          loaded active exited  Raise network interfaces
nginx.service                                               loaded active running A high performance web server and a reverse proxy
raspi-config.service                                        loaded active exited  LSB: Switch to ondemand cpu governor (unless shif
rc-local.service                                            loaded active exited  /etc/rc.local Compatibility
redis-server.service                                        loaded active running Advanced key-value store
rng-tools.service                                           loaded active running rng-tools.service
rpi-eeprom-update.service                                   loaded active exited  Check for Raspberry Pi EEPROM updates
rsyslog.service                                             loaded active running System Logging Service
serial-getty@ttyAMA0.service                                loaded active running Serial Getty on ttyAMA0
ssh.service                                                 loaded active running OpenBSD Secure Shell server
supervisor.service                                          loaded active running Supervisor process control system for UNIX
systemd-fsck-root.service                                   loaded active exited  File System Check on Root Device
systemd-fsck@dev-disk-by\x2dpartuuid-6c586e13\x2d01.service loaded active exited  File System Check on /dev/disk/by-partuuid/6c586e
systemd-journal-flush.service                               loaded active exited  Flush Journal to Persistent Storage
systemd-journald.service                                    loaded active running Journal Service
systemd-logind.service                                      loaded active running Login Service
systemd-modules-load.service                                loaded active exited  Load Kernel Modules
systemd-random-seed.service                                 loaded active exited  Load/Save Random Seed
systemd-remount-fs.service                                  loaded active exited  Remount Root and Kernel File Systems
systemd-sysctl.service                                      loaded active exited  Apply Kernel Variables
systemd-sysusers.service                                    loaded active exited  Create System Users
systemd-tmpfiles-setup-dev.service                          loaded active exited  Create Static Device Nodes in /dev
systemd-tmpfiles-setup.service                              loaded active exited  Create Volatile Files and Directories
systemd-udev-trigger.service                                loaded active exited  udev Coldplug all Devices
systemd-udevd.service                                       loaded active running udev Kernel Device Manager
systemd-update-utmp.service                                 loaded active exited  Update UTMP about System Boot/Shutdown
systemd-user-sessions.service                               loaded active exited  Permit User Sessions
triggerhappy.service                                        loaded active running triggerhappy global hotkey daemon
user-runtime-dir@1000.service                               loaded active exited  User Runtime Directory /run/user/1000
user@1000.service                                           loaded active running User Manager for UID 1000
wpa_supplicant.service                                      loaded active running WPA supplicant

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
```

<br>

\- G.

greiginsydney, 2022-01-15:
'TimeoutStartUSec' vs 'TimeoutStartSec' gets quite a run:

[https://github.com/systemd/systemd/issues/2047](https://github.com/systemd/systemd/issues/2047)

peterjbaylis, 2022-01-15:
Hi Greig;

Two extra services showing on my list. 
I can see celery.service and systemd-timesyncd.service on my service list but they are not showing on your greyBox list.

My diagnostic notes are included here...
***
[celery diagnostics 20220115.txt](https://github.com/greiginsydney/Intervalometerator/files/7874726/celery.diagnostics.20220115.txt)

I think its time for a Coopers Pale Ale, I'll have one for you....

Cheers Pete.

peterjbaylis, 2022-01-15:
Hi Greig;

BTW, at this time the pi is not copying files from the camera or transferring files to my NAS.
I can still ssh to the pi but it is not responding to http requests, so I'm thinking I might try the upgrade process again.
Any thoughts?

Cheers Pete.

.

greiginsydney, 2022-01-15:
Not responding to http requests? But all we've changed was the service file for celery, which isn't involved in the serving of pages.

Does a `sudo /home/pi/setup.sh test` report anything (like nginx?) down?

Try this one and see if it outputs anything: `sudo systemctl list-units --failed`

A fresh upgrade will update any of the files that have changed along the way, but it's not as comprehensive as a full rebuild/reinstall. (Don't forget the critical 'cp ~/www/intvlm8r.py ~/www/intvlm8r.old' line).

\- G.

greiginsydney, 2022-01-15:
Hi Pete,

A few more bits:

> daemon reload and show celery, note the 'U' has found its way back to the TimeoutStart parameter.

From what I've been reading, you specify without the U, but when you query, it adds the U. Check out that long issue from systemd that I posted yesterday. So don't be worried by this.

BUT on that, it might be worthwhile pulling out the U in the TimeoutStopUSec entry - or removing it altogether from the celery.service file (as it's redundant specifying the default value).

> I can see celery.service and systemd-timesyncd.service on my service list but they are not showing on your greyBox list.

I can't explain celery's absence from my earlier post, as it's present when I rerun the command now. As for timesyncd, its presence will depend on whether you're using NTP or the Arduino as your master time source. That will be output as the last line of the 'test' command. Here's mine:
```
NTP is not active. The Arduino is our master time source.
```
... whilst yours is
```
NTP is currently active. NTP is our master time source.
```

peterjbaylis, 2022-01-15:
Hi Greig;

I was bit too impatient waiting for the web services to start. It did
eventually start talking to me.
I’ll look at the other tests you suggested and let you know what I find.

Thanks for your patience following this through.

Cheers Pete.

peterjbaylis, 2022-01-17:
Hi Greig;

The pi is behaving today. 
cameraTransfer copied 600 files to the pi and the pi transferred them all to the NAS with no errors.

The only change I made since I last updated this issue is that I formatted the memory card in the camera after manually copying the files to my PC.
I did not update the uploadedOK.txt.

I have not upgraded or updated the intvlm8r installation on the pi SD card. 
I may still do that after I pick up a new microSD card. But I will let it run for a few days with the current image.
And this time I will save the intvlm8r.py to preserve my password.

Cheers Pete.

peterjbaylis, 2022-01-17:
Hi Greig;

The list-units failed says "0 loaded units listed" 

And here is the output from the setup.sh test command.
***
pi@picam-03:~ $ sudo /home/pi/setup.sh test
TEST!
PASS: /etc/nginx/sites-available/intvlm8r exists
PASS: /etc/systemd/system/intvlm8r.service exists
PASS: /etc/systemd/system/cameraTransfer.service exists
PASS: /etc/systemd/system/piTransfer.service exists
PASS: /etc/systemd/system/setTime.service exists
PASS: /etc/systemd/system/redis.service exists
PASS: /etc/systemd/system/celery.service exists
PASS: i2c_arm_baudrate is present in /boot/config.txt
PASS: i2c-dev installed in /etc/modules

If the Arduino is connected & programmed it will show as "04" in the top line below:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- 04 -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

PASS: The Pi is NOT an AP
PASS: WiFi power save is OFF

PASS: nginx    service is running
PASS: intvlm8r service is running
PASS: celery   service is running
PASS: redis    service is running
PASS: remoteit service is not installed

Service = setTime
  LoadState   = loaded
  ActiveState = inactive
  SubState    = dead

Service = cameraTransfer
  LoadState   = loaded
  ActiveState = inactive
  SubState    = dead

Service = piTransfer
  LoadState   = loaded
  ActiveState = inactive
  SubState    = dead

Service = heartbeat
  LoadState   = loaded
  ActiveState = active
  SubState    = waiting

NTP is currently active. NTP is our master time source.
***

Cheers Pete.

greiginsydney, 2022-01-17:
Hi Pete,

I've just created a fresh issue for me to log improvements to the monitoring of Celery.

If yours is behaving now, please feel free to close this issue.

peterjbaylis, 2022-01-18:
Hi Greig;

Thanks for your follow through with this issue.

Cheers Pete.

closed 2022-01-18 by peterjbaylis