greiginsydney/Intervalometerator #113: Is it possible to Upload each image immediately after taking a shot? Need to use Sim card for data.

NikkhhilThackar, 2022-02-10:
Hi Greig

First, thank you for this wonderful project. I have ordered PCB from you and now started building it.
I have 2 questions.
1. Is it possible to upload each image immediately after taking a shot?
2. How can I use a sim card for data instead of Wifi?

Regards,

greiginsydney, 2022-02-10:
Hi [Nikkhhil](https://github.com/NikkhhilThackar), welcome aboard!

I'm glad your PCB made it to you OK. To answer your questions:

>  Is it possible to upload each image immediately after taking a shot?

No, the intvlm8r 'batches' this, and the UI is implemented so this happens once a day. You _can_ make manual changes that will do this multiple times a day, and I've documented in the [FAQ ](https://github.com/greiginsydney/Intervalometerator/blob/master/docs/FAQ.md#can-i-copytransfer-images-more-than-once-per-day) how to make the copy/transfer take place once an hour.

The intvlm8r is designed to run from solar and battery, and you'd need to have the Pi running permanently to support the transfer of every image as it's taken.

There's an extra layer of complication in your request because the Pi would need to know *when* an image has been taken so it can immediately copy it out of the camera, and there's no way in the hardware for the Arduino (which manages the shooting schedule) to signal this to the Pi. This leaves me with the Pi needing to wake and poll the camera potentially every minute looking for new photos...

>  How can I use a sim card for data instead of Wifi?

Plenty of people have done this, and all you need is a "4G/5G WiFi dongle". It gives you the network connectivity and makes a WiFi network for the intvlm8r to connect to.  "remote.it" is a popular service that can let you access your intvlm8r in the field over such a connection. The downside of this is the power requirements of the dongle, which again is problematic if you're running off solar/battery and also want a frequent upload of images.

\- Greig.

NikkhhilThackar, 2022-02-11:
Thank you Greig for the quick response. Will look deeper into this.

closed 2022-03-19 by greiginsydney