greiginsydney/Intervalometerator #114: Transfer of images not taking place

aerialvideographyuk, 2022-02-25:
Have done a complete new install of the latest arduino code and pi code and everything worked as expected.
Installed remoteit and no transfer took place.  I remember we had to do a change to make celery wait until remoteit had loaded .  It looks like the way remoteit installs is now different and I couldn't find how to replicate the previous fix.
Here is the result of running setup.sh test.
Although it says remoteit service is not installed, it is installed.

After some further investigation I found the remoteit@.service  in /etc/systemd/system and did the same change as previous
After=network.target rc-local.service celery.service but celery still did not start.
Noticed also that I don't see 04 in the top line below even though the arduino appears to be working OK?
Any help would be very welcome.

pi@avuk1:~ $ sudo /home/pi/setup.sh test
TEST!
PASS: /etc/nginx/sites-available/intvlm8r exists
PASS: /etc/systemd/system/intvlm8r.service exists
PASS: /etc/systemd/system/cameraTransfer.service exists
PASS: /etc/systemd/system/piTransfer.service exists
PASS: /etc/systemd/system/setTime.service exists
PASS: /etc/systemd/system/redis.service exists
PASS: /etc/systemd/system/celery.service exists
PASS: i2c_arm_baudrate is present in /boot/config.txt
PASS: i2c-dev installed in /etc/modules

If the Arduino is connected & programmed it will show as "04" in the top line below:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

PASS: The Pi is NOT an AP
PASS: WiFi power save is OFF

PASS: nginx    service is running
PASS: intvlm8r service is running
FAIL: celery   service is dead
PASS: redis    service is running
PASS: remoteit service is not installed

Service = setTime
  LoadState   = loaded
  ActiveState = inactive
  SubState    = dead

Service = cameraTransfer
  LoadState   = loaded
  ActiveState = inactive
  SubState    = dead

Service = piTransfer
  LoadState   = loaded
  ActiveState = inactive
  SubState    = dead

Service = heartbeat
  LoadState   = loaded
  ActiveState = active
  SubState    = waiting

NTP is not active. The Arduino is our master time source.

pi@avuk1:~ $

greiginsydney, 2022-02-25:
Hi Dave,
<br>

> Noticed also that I don't see 04 in the top line below even though the arduino appears to be working OK?

Yeah, this seems to be a thing in Bullseye. I've been able to reproduce this here myself and I'm tracking it as Bug #112. I can't find anyone else reporting similar issues, but as Bullseye (on the Pi) is relatively new I'm hoping a resolution might arrive sometime soon.

> Although it says remoteit service is not installed, it is installed.

Great, thanks for letting me know. I'll do a fresh install and review/refresh the 'installed' test.

> FAIL: celery service is dead

Given past history I'm guessing remote.it is killing Celery, so let me look some more into that.

Is [this the new install process](https://support.remote.it/hc/en-us/articles/4422773654669-Streamlined-installation-for-Linux-and-Raspberry-Pi-platforms) you followed?

\- G.

aerialvideographyuk, 2022-02-25:
Hi Greig,

 

I just ran the following command on the pi, as it seemed that curl was already installed.

 

sh -c "$(curl -L https://downloads.remote.it/remoteit/install_agent.sh)"

 

That generated a code to put into the desktop application and that seemed to work fine and enabled me to connect via ssh and http.

 

Prior to the remoteit install I had checked all functions were working including the rename on copy and dead mans snitch.

Just for completeness I also remove files from the camera when copied and from the pi when transferred.

I am transferring to a dropbox account and then have a local script that tranfers to a NAS at night.

 

I haven’t as yet uninstalled remoteit as I felt it may be best to leave it on to aid fault finding.

I can collect the data from the camera’s sdcard as it is local to me at the moment.

 

I will be moving my web hosting soon and so have put a personal mail address in above in case there is an interruption to my aerialvideographyuk.com mail.

 

All the best,

Dave

 

Hi Dave,

This email has been checked for viruses by AVG.
https://www.avg.com

greiginsydney, 2022-02-25:
Aha!

It looks like they've changed the name of the service to `connectd.service`, so if you try adding that "After=" edit to THIS file instead, you might be in luck.

```bash
/etc/systemd/system/connectd.service
```
<br>

\- G.

aerialvideographyuk, 2022-02-26:
I added the celery.service to the After= for the schannel.service and that didn't cure the problem.
I then also added it to the connectd.service and that seems to have fixed it.
Should I remove  the entry from schannel.service or is it ok to leave it there?

greiginsydney, 2022-02-27:
> I then also added it to the connectd.service and that seems to have fixed it.

Thanks for confirming that. My analysis suggested it would be the other, so I appreciate the feedback. I'll update the doco.

> Should I remove the entry from schannel.service or is it ok to leave it there?

Safest to remove it, thanks.

\- G.

greiginsydney, 2022-03-19:
I've had no luck engaging remote.it, but it looks like the service is in a state of flux. (The above changes seem to no longer work!)

I'll keep an eye on it and update the code in a future release.

greiginsydney, 2022-03-23:
Support @ remote.it has responded and provided a better way of detecting its presence, which I'll add into the setup script.

I'm hoping they'll be able to clarify the service file(s) we need to edit so they'll play well with celery.

greiginsydney, 2022-05-21:
Fixed in [v4.4.3](https://github.com/greiginsydney/Intervalometerator/releases/tag/v4.4.3) released 21st May 2022

closed 2022-05-21 by greiginsydney