greiginsydney/Intervalometerator #116: Preview images not included in daily download count.

peterjbaylis, 2022-04-14:
Hi Greig;

If I ask the Intvlm8r to take a preview shot during the normal active schedule then those shots do not get transferred to the pi and subsequently to my FTP server.
Is this intentional  i.e. is it supposed to work this way?
I would like to see an option to allow these additional “Preview” images to be included in the daily piTransfer process.

Cheers Peter.

greiginsydney, 2022-04-15:
Hiya Peter,

You're seeing design intent; yes, it's supposed to work this way.

Preview images are disposable ones used in the camera setup process, and what you're seeing isn't actually the "real" captured image, but what's essentially a glorified "thumbnail". [[Reference]](https://photo.stackexchange.com/questions/99936/is-the-preview-file-always-the-photo-taken-by-the-camera)

The camera doesn't save them to the memory card - they're instead transferred 'live' over USB and then I display them in the page.

You can keep a copy of a Preview image by clicking on it, which will open it in a new browser tab, then right-click to save.

\- G.

peterjbaylis, 2022-04-15:
Greig;

Ok, thanks for the explanation.

Cheers Pete.

closed 2022-04-15 by peterjbaylis