greiginsydney/Intervalometerator #123: Canon camera intermittently reports 'busy' on LCD & won't take photos

greiginsydney, 2022-08-20:
Some models of Canon camera may manifest an intermittent issue where the camera's LCD display reports "busy", and the camera won't take a photo.

Unplugging the USB cable restores the ability to take photos.

<p align="center">
[image]
 </p>

This appears to be a bug in the version of libgphoto (one of the underlying components) that's installed with the "bullseye" version of Raspberry Pi o/s. [[Ref#1](https://github.com/gphoto/libgphoto2/issues/352)] [[Ref#2](https://github.com/gphoto/gphoto2/issues/465)]

Intvlm8r's still running on the earlier "buster" o/s are not affected.

### How to tell if you're at risk

From the "System Maintenance" page, check the `Pi o/s version` and `libgphoto2 version`:

<p align="center">
[image]
 </p>

__libgphoto2 version 2.5.27 is the problem.__

### Resolution

An update to the `setup.sh` script is currently being soak-tested. If it detects a "legacy" installation of libgphoto2 it will first uninstall it and then install the latest version directly from GitHub.

An upside to this change is that newer camera models will be supported by the intvlm8r sooner than would have otherwise been the case.

<br>

\- Greig.

wj7106, 2022-08-21:
Last year, I reflected on this problem. Maybe the description was not very clear at that time, https://github.com/greiginsydney/Intervalometerator/issues/80

greiginsydney, 2022-08-21:
Nah, that was a different scenario. In this case all USB-related comms is fine: you can take and view a Preview image and view/change all other camera parameters, however the on-camera 'shoot' button is inoperative, as is the remote trigger input.

greiginsydney, 2022-09-24:
Fixed in Release v4.5.0.

closed 2022-09-24 by greiginsydney