greiginsydney/Intervalometerator #136: Save to USB

DaveJ1013, 2023-05-24:
Hi Greig,   have a potential customer that wont allow wifi on the site. Was wondering about the possibility of saving the pics to a USB stick and replacing the USB stick with a new one every few weeks. 

Cheers
Dave Jeffries

greiginsydney, 2023-05-25:
Hi Dave,

Ouch. Tough ask.

How strict is this customer? Is the Wi-Fi radio itself banned from site, or just the use of? (I've worked in an environment before where devices couldn't even possess the radio hardware).

The Pi Zero "non W" that would meet the former requirement, but that looks to be the only recent (current) Pi without Wi-Fi, and without an Ethernet port you couldn't get to the web UI to manage it. (I'm avoiding going down the rabbit-hole of USB hubs and USB Ethernet adapters).

Assuming we're allowed a Pi with a W-Fi radio that's disabled (including having removed all its drivers), and Ethernet, the trick would be to connect either a memory stick or an external SSD to one of the Pi's USB ports, and redirect the /photos/ directory to that location. It'd need to handle you removing/replacing that drive, but it wouldn't surprise me if Raspbian already does that.

Of course if Ethernet's allowed - at least locally on site - you have management access and it can upload the images directly to a PC/laptop each day.

Another "solution" not without risk is to just use a giant memory card in the camera, but each time you replace it you risk moving the enclosure or camera... And that's a single point of failure.

<br>

\- Greig.

DaveJ1013, 2023-05-25:
Hi Greig,They are OK with it containing WiFi as long as it does not connect to any external network, including via 4g.  The cameras would be on existing towers (about 20 mtrs high) and have to be solar powered.  Access to the cameras afterwards could only be achieved with a hydraulic platform.  Hence the idea of an external USB or SSD on a long wire.I am hoping that a once a month visit for data collection would work.  Size of data shooting raw at 120 pictures a day would be around 80Gb a month for each of 2 cameras  Even if I could connect via WiFi at the visit,  I hate to think how long the transfer would take!The site will be a munitions factory and the build is scheduled to take around 2 years.  Any other ideas would be very welcome.All the bestDave JFrom: Greig Sheridan ***@***.***>Sent: Thursday, 25 May 2023, 02:14To: greiginsydney/Intervalometerator ***@***.***>Cc: DaveJ1013 ***@***.***>; Author ***@***.***>Subject: Re: [greiginsydney/Intervalometerator] Save to USB (Issue #136)
Hi Dave,
Ouch. Tough ask.
How strict is this customer? Is the Wi-Fi radio itself banned from site, or just the use of? (I've worked in an environment before where devices couldn't even possess the radio hardware).
The Pi Zero "non W" that would meet the former requirement, but that looks to be the only recent (current) Pi without Wi-Fi, and without an Ethernet port you couldn't get to the web UI to manage it. (I'm avoiding going down the rabbit-hole of USB hubs and USB Ethernet adapters).
Assuming we're allowed a Pi with a W-Fi radio that's disabled (including having removed all its drivers), and Ethernet, the trick would be to connect either a memory stick or an external SSD to one of the Pi's USB ports, and redirect the /photos/ directory to that location. It'd need to handle you removing/replacing that drive, but it wouldn't surprise me if Raspbian already does that.
Of course if Ethernet's allowed - at least locally on site - you have management access and it can upload the images directly to a PC/laptop each day.
Another "solution" not without risk is to just use a giant memory card in the camera, but each time you replace it you risk moving the enclosure or camera... And that's a single point of failure.

- Greig.

greiginsydney, 2023-05-25:
OK, so if the intvlm8r can connect to an "internal" Wi-Fi network that's on site, I'd drop a [conveniently ground-level] 'management server' there, and all the intvlm8r's would upload on their schedule to *it*. The could be a Windows PC or even another Pi.

Operating as an island like that you'd have no heartbeating, but with two cameras each provides a degree of redundancy should the other take an unexpected break from shooting.

Unrelated question: what timeframe are you looking at? The real-time clock IC is only now *just* returning from 'unobtainium'. I can buy it from component suppliers overseas today (news to me just now actually!), but the plugin Sparkfun unit spec'd for the intvlm8r isn't yet shipping. It shouldn't be won't be long though.

\- G.

DaveJ1013, 2023-05-25:
Hi Greig,I'll ask some more questions to check my understanding but, I don't think it will be possible to locate a server as there's no power anywhere near the camera locations.  It's going to be a few months until ground work starts, but I have 1 camera ready to go and a spare board already built.  I could build the 2nd camera up in a few days if needed.All the bestDave JFrom: Greig Sheridan ***@***.***>Sent: Thursday, 25 May 2023, 08:50To: greiginsydney/Intervalometerator ***@***.***>Cc: DaveJ1013 ***@***.***>; Author ***@***.***>Subject: Re: [greiginsydney/Intervalometerator] Save to USB (Issue #136)
OK, so if the intvlm8r can connect to an "internal" Wi-Fi network that's on site, I'd drop a [conveniently ground-level] 'management server' there, and all the intvlm8r's would upload on their schedule to it. The could be a Windows PC or even another Pi.
Operating as an island like that you'd have no heartbeating, but with two cameras each provides a degree of redundancy should the other take an unexpected break from shooting.
Unrelated question: what timeframe are you looking at? The real-time clock IC is only now just returning from 'unobtainium'. I can buy it from component suppliers overseas today (news to me just now actually!), but the plugin Sparkfun unit spec'd for the intvlm8r isn't yet shipping. It shouldn't be won't be long though.
- G.

greiginsydney, 2023-05-27:
Ok, so assuming Wi-Fi is permissible, I'd:
- Build the PCB and camera in a box, mounted high up the pole
- Put a GIANT memory card in the camera, and same in the Pi
- Locate the battery, charger and possibly the solar cell in/on a box at or near ground level
- Extend the reed switch cabling to the lower box. This will allow you to wake the intvlm8r on demand. (You could do this crudely by simply putting a cutoff (key?) switch in the power line up to the camera, as the Pi will always wake upon the re-connection of power. Only caveat here is that you DON'T want to be cutting power when the Pi is running, as that risks drive corruption. I've never had it happen, but Murphy's Law...)

When woken, the Pi will come up as an access point and you can connect your PC/Mac to it and suck the images, either [manually](https://github.com/greiginsydney/Intervalometerator/blob/bugs2304/docs/manually_download_images.md), or by firing a "Transfer Now" from the /transfer page, with your computer setup as the target, running ftp, rsync, etc.

How's that sound?

If you can't Wi-Fi things get "interesting".

DaveJ1013, 2023-05-27:
Hi Greig,Will give this a try with a couple of 128Gb cards.  I was thinking of leaving all the power,  solar and charger on the mast and just bringing the reed switch out. Any issue you see with that?Would there be any difference in using my laptop as the access point so that the normal setup stays on the intvlm8 as opposed to the AP configuration? My only fear is how long the transfer will take!  All the bestDave JFrom: Greig Sheridan ***@***.***>Sent: Saturday, 27 May 2023, 07:22To: greiginsydney/Intervalometerator ***@***.***>Cc: DaveJ1013 ***@***.***>; Author ***@***.***>Subject: Re: [greiginsydney/Intervalometerator] Save to USB (Issue #136)
Ok, so assuming Wi-Fi is permissible, I'd:

Build the PCB and camera in a box, mounted high up the pole
Put a GIANT memory card in the camera, and same in the Pi
Locate the battery, charger and possibly the solar cell in/on a box at or near ground level
Extend the reed switch cabling to the lower box. This will allow you to wake the intvlm8r on demand. (You could do this crudely by simply putting a cutoff (key?) switch in the power line up to the camera, as the Pi will always wake upon the re-connection of power. Only caveat here is that you DON'T want to be cutting power when the Pi is running, as that risks drive corruption. I've never had it happen, but Murphy's Law...)

When woken, the Pi will come up as an access point and you can connect your PC/Mac to it and suck the images, either manually, or by firing a "Transfer Now" from the /transfer page, with your computer setup as the target, running ftp, rsync, etc.
How's that sound?
If you can't Wi-Fi things get "interesting".

greiginsydney, 2023-05-27:
> I was thinking of leaving all the power,  solar and charger on the mast and just bringing the reed switch out. Any issue you see with that?

My primary concern there is with the camera 20m up in the air, I'd be happier having the power side of things more accessible to you at ground level.

My friend @techydude has also wisely counselled against extending the reed line down the pole, as it will become a great antenna for injecting ESD into the board. I'd instead employ a relay or opto-isolator to electrically isolate the cable coming down the pole from the PCB. Let me know if that's not clear and I'll rough up a circuit.

> Would there be any difference in using my laptop as the access point so that the normal setup stays on the intvlm8 as opposed to the AP configuration? 

I don't think you'll get your lappy to make an AP, so you'll need to bring an independent AP with you that the intvlm8r and lappy will join its network.

> My only fear is how long the transfer will take!

Yes, it might take a while depending on how frequently you visit the site.

In an off-grid setup the Pi will only power up for the maximum of an hour before it goes back to sleep, but with you on-site you can keep pressing the 'reed' button and that will extend the timer for another hour each time.

closed 2023-06-28 by greiginsydney