greiginsydney/Intervalometerator #137: Replacement for DeadOn RTC: DS3231?

franioli, 2023-05-27:
Hi Greig,
Thanks for your great work!
I've already built 2 intervalometers for my PhD project on high-frequency glacier monitoring using stereo photogrammetry. The two cameras have been working for more than a year on a glacier in the Italian Alps. 
Now I was thinking of building another camera to extend the monitored area, but the RTC clock seems to be impossible to get... So I wanted to ask you if you think there are any alternatives that could work without a massive rewrite of the Arduino code. I found that the closest might be the DS3231, which should be almost as accurate as the Sparkfun and supports two alarms, but has an I2C interface. Do you have any experience with this? Or do you have any suggestions? 

Thanks again! 

Francesco

greiginsydney, 2023-05-27:
Hi Francesco,

It's great to hear the intvlm8r has been of use to you, and that the boards have such a lovely view!

As you've already found, I painted myself into a corner using the Deadon RTC boards, as that clock IC is the only one on the planet that uses an SPI interface. Unfortunately I don't have the option of rewriting the code for I2C because of how the Pi and Arduino use that bus.

BUT THERE'S GOOD NEWS!

After a couple of years out of stock, the DS3234 is on its way back! SparkFun doesn't seem to have them yet, but that can't be far off, as Mouser has them (the chip) available today.

Do you have a deadline for when you need this new unit?

\- G.

franioli, 2023-05-29:
Hi Greig, 
thanks for the answer. I can understand your point.
You're right about the chips, but actually Mouser says that the expected date for the RTC availability will be in Dicember 2023. That's a bit too late for me, as I would like to build a new system before winter, but that's it... let's wait and hope in good news.

Btw, I am currently using SIM7600E-H 4G HAT (https://www.waveshare.com/wiki/SIM7600E-H_4G_HAT) connected to the Rasp Zero for 4G connection. If you are interested, I can try to write a piece of documentation and add it in the setup-remote guide.

greiginsydney, 2023-05-29:
They have 1,717 of the [DS3234S#](https://au.mouser.com/ProductDetail/Analog-Devices-Maxim-Integrated/DS3234S?qs=1eQvB6Dk1vhNRvcRKqeJUg%3D%3D), which is the 0-70°C version - the same as what SparkFun uses.

I guess yes if you'll be sitting atop a glacier, the DS3234SN# rated for -40°C to +85°C would be a better idea. Element14 says they're "Awaiting Delivery by 22/08/23", which is a more friendly date.

If you're up for some surface-mount soldering (or know someone who can do it), I might be able to find you a prototype PCB that will take the 3234SN directly on-board.

And thanks for the offer of the doco edit. I'd be delighted if you could branch the repo and add the SIM7600E-H 4G HAT as a new heading after "4G/5G dongle" (immediately before "install remote.it"). Some images and text would be lovely. Don't worry about formatting - I can do that. Let me know if that's a blur of techno-babble and I can walk you through it.

\- G.

closed 2023-06-28 by greiginsydney