greiginsydney/Intervalometerator #138: Function suggestion: Add flash hot shoe signal to arduino

wj7106, 2023-05-27:
Suggestion: Can the shutter end signal from the camera flash hot shoe be introduced into Arduino to count the number of successful shots taken on that day or month in the system? I have seen that the Pixhawk PIX4 used in the aircraft model has this module function, which can count the number of successful shots. If the intvlm8r can add this input function, I think I can fully use ultra high definition 8k records like Sony a7rm2!

greiginsydney, 2023-05-27:
That's a neat idea.

What are you proposing the Arduino would do if the 'fire' signal doesn't see a corresponding flash? Obviously a re-take attempt would be the first action, but then what?

How often does the intvlm8r trigger a shot that the camera fails to take? I can only see that happening if the camera's in AF mode, and the trigger time isn't long enough. 

<br>

\- G.

wj7106, 2023-05-28:
By using the flash hot shoe signal, the camera's shooting results can be more directly monitored. The number of shots taken and missed shots will be clear at a glance. For some cameras that do not support the gphoto plugin, such as the Sony A7 series, Fuji series, or other micro single camera, as long as the camera with the flash hot shoe is equipped, except for not being able to connect to raspberry zero via USB to preview and transfer photos to the pi, it can be controlled by the shutter cable for shooting. I have been using this system to shoot engineering projects for 2 years, so I don't need to record the current number of shots every time I log in to the system