greiginsydney/Intervalometerator #139: Camera time not updating from the web interface

aerialvideographyuk, 2023-05-27:
Just put a test camera up for a future project.
While setting the time from the web interface for pi, arduino and camera, the camera doesn't set and goes to 2000 Jan 01 00:00:00
Further attemps at just setting the camera only results in the same issue.

greiginsydney, 2023-05-27:
Hi Dave,

<br>

"Alt camera time mode" should fix that:

<br>

[image]

Let me know if not and hopefully we can work together on a better fix.

<br>

\- G.

greiginsydney, 2023-06-01:
Hey @aerialvideographyuk have you had any luck with that?

greiginsydney, 2023-06-16:
I've created a new FAQ to help anyone encountering this issue in the future:
-  [Why cant I set the camera's time correctly](https://github.com/greiginsydney/Intervalometerator/blob/master/docs/FAQ.md#Why-cant-I-set-the-cameras-time-correctly)

closed 2023-08-16 by greiginsydney