greiginsydney/Intervalometerator #142: Function suggestion: Add to Intervalometer Settings Interval in Seconds.

szesnascie, 2023-06-20:
Sometimes I need to take pictures more than every minute, sometimes even every few seconds. It would be nice if you could add an additional option in which you could choose the interval in seconds.

greiginsydney, 2023-06-21:
Hi @szesnascie, thanks for the suggestion.

Can you tell me a little more about your application? You need all the functionality of the intvlm8r *and* potentially as many as 20 or 30 images a minute through each day's shooting hours?

Ramping up the shooting schedule to that frequency would not be without its challenges, but I think the bigger issue would then become getting all those images off the camera, then off the Pi.

\- G.

szesnascie, 2023-06-21:
hi, @greiginsydney I often have a situation that I need to register a process that lasts an hour, maximum two hours. The current configuration allows me to set a maximum of 60 photos in an hour, of which I can get 2s of film, which is definitely not enough so I have to show up with my equipment on site and record it separately. Most often, these are situations that occur suddenly because for some reason the work schedule has been changed. Thanks to this functionality, I could remotely manage a few hours schedule that would very accurately record the process I am interested in. After the end, I would change the schedule to "normal intervals".

greiginsydney, 2023-06-22:
So you have an intvlm8r up covering a 'build' of some sort that has bursts of intense activity that you need to capture?

I presume this is a site with power, so you have the luxury of the Pi always running, and the ability to update the schedule remotely at any time?

If I were able to update the shooting frequency to 'every x seconds', what's the minimum 'x' (maximum 'frame rate') you'd be seeking? (I'm guessing during these bursts that would be the one rate you'd shoot at? You'd go from 'normal' intervals, to the fastest possible, then back again later?)

greiginsydney, 2023-06-22:
If you're shooting a hypothetical 10 shots a minute for two hours, that's 1200 shots, and at 5MB each, you've shot 6 Gig you need to get off the camera to the Pi, then uploaded somewhere... Is this do-able without a site visit anyway? And how often does this happen?

szesnascie, 2023-06-23:
Exactly, I have power all the time. The minimum 'x' is 5. File management is not a problem for me. I have this event once a week, sometimes less often.

greiginsydney, 2023-06-24:
OK, leave it with me and I'll peruse the code and see how readily it could be implemented - and if doing so won't upset other functionality.

\- G.

szesnascie, 2023-06-24:
🙏🏻🙌🏻🙌🏻🙌🏻

szesnascie, 2023-07-03:
Hi, @greiginsydney did you have a time to peruse the code and see how readily it could be implemented?

-Sz.

greiginsydney, 2023-07-03:
Hi @szesnascie, I have and I think it's something the Arduino can do.

My current plan is for the timer to fire each minute - as it does now when set for 60 shots/hour - but each firing triggers a sequence of timed shots rather than just one. The outcome is a shot every 'n' seconds.

I'm in NZ at the moment but when I'm home next week I'll rough this up in the lab and see how reliably it works.

(I have an update in the wings that will be released before I start on this).

I'll let you know how it goes.

\- Greig.

szesnascie, 2023-07-03:
@greiginsydney awsome great news, can't wait! Enjoy your stay!!!
- Sz.

greiginsydney, 2023-07-16:
A first test of the code today. Rather than work the camera's shutter for these test shots, I hook my digital counter up.

This is what an HOUR of shooting results in when you set a 3s interval!

[image]

(I'm still not sure I want to let this out the door shooting at that frequency, but that's a number I'm toying with at the mo).

\- Greig.

szesnascie, 2023-07-16:
Awsome Greig.
We can use one of my stations for testing before release.
Sz

greiginsydney, 2023-07-16:
This update requires changes to both the Arduino and Pi. Do you have a [programming cable for the Arduino](https://www.sparkfun.com/products/9873)?

greiginsydney, 2023-10-29:
This functionality will be added in [v4.6.0](https://github.com/greiginsydney/Intervalometerator/releases/v4.6.0), due this side of Christmas. 

You'll be able to shoot up to one image every **three seconds**.

This mode comes with a big "at your own risk" caveat, and at its fastest probably requires a supervised environment where you have mains power and direct access to the camera to replace the memory card at the end of each day's shooting.

Here's a sneak-peek at the UI:

[image]

<br>

[image]

szesnascie, 2023-10-30:
This is great news, thank you for your work and time spent on the project!

greiginsydney, 2023-12-10:
Closed with the release of [v4.6.0](https://github.com/greiginsydney/Intervalometerator/releases/tag/v4.6.0)

closed 2023-12-10 by greiginsydney