greiginsydney/Intervalometerator #144: Remote.it installs OK, but setup.sh fails with 'remoteit is not installed' & celery no longer starts

greiginsydney, 2023-06-28:
Remote.it installs OK:

[image]

... but the install script doesn't detect it:

[image]

The [remote.it] "connectd" service no longer exists.

Remote.it has changed their packaging for the Pi, and the "connectd.service has not existed for quite a while."

greiginsydney, 2023-06-28:
[image]

Celery *initially* fails to start, but does eventually.

mmunsk, 2023-10-22:
Hi, does this mean that the intervalometerator doesn't work anymore? How does this affect its functionality?

greiginsydney, 2023-10-22:
All good @mmunsk - turns out it's all good.

With its new install process, remote.it - and the intvlm8r - work fine, it's just that my previous fix is now redundant.

Because it all works so well I've not made it a priority to update the './setup test' code. Ta for the nudge.

\- G.

mmunsk, 2023-10-23:
Hey Greig,
Quick question.
I'm setting up a camera far away.
If I needed to get heartbeats can I just have them sent to my email?
Or is this just for a website URL?

In any case how can I set this up to get these heartbeats and make sure the camera is working?

Thanks.

greiginsydney, 2023-10-23:
Check out the [setup-heartbeating.md](https://github.com/greiginsydney/Intervalometerator/blob/master/docs/setup-heartbeating.md) page. There's virtually nothing to do in the intvlm8r itself - it's all in the service you use. I've been using - and love - deadman's snitch.

\- G.

commit cf99ed8f (2023-10-28): Update setup-remote.it.md

Closes #144
closed 2023-10-28 by greiginsydney