greiginsydney/Intervalometerator #156: DeleteAfterCopy=False & DeleteAfterTransfer=True causes unwanted loops

greiginsydney, 2024-02-29:
When I added the DeleteAfterCopy/Transfer features I failed to think of (and test) what would happen if you set DeleteAfterCopy=False & DeleteAfterTransfer=True.

You'd do that in a scenario where you have HEAPS of storage on the camera, but not on the Pi. The copies that make it to the Pi are only there for a brief period then deleted after they've been uploaded. The copies remaining on the camera are your emergency protection.

The problem is that as it stands, the Pi will attempt to copy a growing list of "new" (old) images to the Pi each day...

I need to fix this!

mmunsk, 2024-02-29:
Thanks for following up with this Greig.
This functionality is needed in order to be able to shoot long term projects without the need to delete the files from the limited space on the pi, you could easily have a 1TB SD card in the camera to store all the files as a backup and have ONLY the new files Copied each day from the camera to the pi transfered to the cloud and then immediately deleted from the pi in order to have room permanently.

Another solution could be to add an external ssd hard drive connected to the pi where the files could also be stored and leave the micro sd only as a ROM (operating software only).
Which will also increase the life of that memory card (that could actually get corrupted if a bad file is copied from the camera among other potential risks).

Thanks for the hard work Greig, waiting for one of these solutions.

Cheers,

Tony

greiginsydney, 2024-04-18:
Fixed in [Rls 4.6.3](/releases/tag/v4.6.3)

closed 2024-04-18 by greiginsydney