greiginsydney/Intervalometerator #162: Frequent "502 Bad Gateway" and "Celery not running" warning when home does load

benjigreig, 2025-01-30:
I'm setting up to new boards and putting the current version of the repo on following your docs. I've gotten through the process of getting everything installed and it all seems to be running as intended (based on the instructions).

However after booting the pi, when I log into the intervalometertor local website on the Pi I frequently get 502 errors. If and when I do get the index page I'm generally met with a "Celery not running" warning message.

Not sure how to proceed. Have tried reinstalling from the setup.sh but have the same issue.

benjigreig, 2025-01-30:
Interestingly if I reinstall with the "sudo -E ./setup.sh web" again and then jump into a browser without rebooting it works fine. But post reboot it misbehaves again.

greiginsydney, 2025-01-30:
Celery is probably the slowest component to come up.

Have you tried just waiting a while??

benjigreig, 2025-01-30:
Yeah have left it sit for ages, but it seems to have issues. I've rebuilt on another sd on one of my other boards. Will test that out and clone that SD I think... Fingers crossed.

Will post back once I've tested.

greiginsydney, 2025-01-31:
The Celery log will hopefully reveal the source of its issues:

/var/log/celery/celery_worker.log

greiginsydney, 2025-02-07:
@benjigreig have you had any luck with the Celery log?

Is this problem related to just one board, or ALL the new ones you've built?

greiginsydney, 2025-02-12:
FYI I've just done a new build of v5.0.0 (admittedly, onto a Pi 3) and it's working fine out of the box. Celery is happy.

(The web setup step threw an error about rc.local, but that isn't an issue, and will be fixed in the next release).

\- G.

benjigreig, 2025-02-12:
OK good to know about the error on rc.local I was seeing that too.
I ended up starting from scratch and managed to get it up and running without the issues recurring.
Will have to chalk this up to an unknown error on my part. Thanks for rubberducking this with me @greiginsydney.

Benj