greiginsydney/Intervalometerator #163: arduino time?

wj7106, 2025-03-09:
hello, I encountered an issue with Arduino time error. It automatically sets the wrong time every few days of normal operation. I set it to shoot every 60 minutes from 8am to 5pm, and the system set it to start at 12pm every day for 30 minutes. Today, I found out that the shooting time was normal before turning on at 12pm yesterday, taking every 60 minutes. However, when turning on at 12pm, Arduino time changed to 12:25pm and took another shot instead of 13:00

settime.log

2025/02/17 12:00:55 
2025/02/17 12:00:55 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/02/17 12:00:55 Starting new HTTP connection (1): localhost:8080
2025/02/17 12:01:02 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/02/17 12:01:02 Status code = 200
2025/02/17 12:01:02 This is what I received: <p>Set Arduino datetime to 20250217120100</p>
2025/02/17 12:01:02 New Arduino time is 20250217120100
2025/02/18 12:00:53 
2025/02/18 12:00:53 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/02/18 12:00:53 Starting new HTTP connection (1): localhost:8080
2025/02/18 12:01:02 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/02/18 12:01:02 Status code = 200
2025/02/18 12:01:02 This is what I received: <p>Set Arduino datetime to 20250218120100</p>
2025/02/18 12:01:02 New Arduino time is 20250218120100
2025/02/19 12:00:52 
2025/02/19 12:00:52 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/02/19 12:00:52 Starting new HTTP connection (1): localhost:8080
2025/02/19 12:01:01 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/02/19 12:01:01 Status code = 200
2025/02/19 12:01:01 This is what I received: <p>Set Arduino datetime to 20250219120059</p>
2025/02/19 12:01:01 New Arduino time is 20250219120059
2025/02/20 12:00:54 
2025/02/20 12:00:54 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/02/20 12:00:54 Starting new HTTP connection (1): localhost:8080
2025/02/20 12:01:02 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/02/20 12:01:02 Status code = 200
2025/02/20 12:01:02 This is what I received: <p>Set Arduino datetime to 20250220120100</p>
2025/02/20 12:01:02 New Arduino time is 20250220120100
2025/02/21 12:00:55 
2025/02/21 12:00:55 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/02/21 12:00:55 Starting new HTTP connection (1): localhost:8080
2025/02/21 12:01:03 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/02/21 12:01:03 Status code = 200
2025/02/21 12:01:03 This is what I received: <p>Set Arduino datetime to 20250221120101</p>
2025/02/21 12:01:03 New Arduino time is 20250221120101
2025/02/22 12:00:55 
2025/02/22 12:00:55 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/02/22 12:00:55 Starting new HTTP connection (1): localhost:8080
2025/02/22 12:01:09 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/02/22 12:01:09 Status code = 200
2025/02/22 12:01:09 This is what I received: <p>Set Arduino datetime to 20250222120107</p>
2025/02/22 12:01:09 New Arduino time is 20250222120107
2025/02/23 12:00:53 
2025/02/23 12:00:53 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/02/23 12:00:53 Starting new HTTP connection (1): localhost:8080
2025/02/23 12:01:02 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/02/23 12:01:02 Status code = 200
2025/02/23 12:01:02 This is what I received: <p>Set Arduino datetime to 20250223120100</p>
2025/02/23 12:01:02 New Arduino time is 20250223120100
2025/02/23 12:32:50 
2025/02/23 12:32:50 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/02/23 12:32:50 Starting new HTTP connection (1): localhost:8080
2025/02/23 12:32:52 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/02/23 12:32:52 Status code = 200
2025/02/23 12:32:52 This is what I received: <p>Set Arduino datetime to 20250223123250</p>
2025/02/23 12:32:52 New Arduino time is 20250223123250
2025/02/25 11:28:57 
2025/02/25 11:28:57 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/02/25 11:28:57 Starting new HTTP connection (1): localhost:8080
2025/02/25 11:29:06 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/02/25 11:29:06 Status code = 200
2025/02/25 11:29:06 This is what I received: <p>Set Arduino datetime to 20250225112904</p>
2025/02/25 11:29:06 New Arduino time is 20250225112904
2025/02/26 12:00:52 
2025/02/26 12:00:52 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/02/26 12:00:52 Starting new HTTP connection (1): localhost:8080
[... 63 lines cut ...]
2025/03/06 12:01:03 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/03/06 12:01:03 Status code = 200
2025/03/06 12:01:03 This is what I received: <p>Set Arduino datetime to 20250306120101</p>
2025/03/06 12:01:03 New Arduino time is 20250306120101
2025/03/07 12:00:55 
2025/03/07 12:00:55 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/03/07 12:00:55 Starting new HTTP connection (1): localhost:8080
2025/03/07 12:01:03 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/03/07 12:01:03 Status code = 200
2025/03/07 12:01:03 This is what I received: <p>Set Arduino datetime to 20250307120100</p>
2025/03/07 12:01:03 New Arduino time is 20250307120100
2025/03/07 12:35:41 
2025/03/07 12:35:41 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/03/07 12:35:41 Starting new HTTP connection (1): localhost:8080
2025/03/07 12:35:43 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/03/07 12:35:43 Status code = 200
2025/03/07 12:35:43 This is what I received: <p>Set Arduino datetime to 20250307123541</p>
2025/03/07 12:35:43 New Arduino time is 20250307123541
2025/03/08 12:34:53 
2025/03/08 12:34:53 systemd-timesyncd = active. The Pi takes its time from NTP. Updating the Arduino with time from the Pi
2025/03/08 12:34:53 Starting new HTTP connection (1): localhost:8080
2025/03/08 12:34:55 http://localhost:8080 "GET /setArduinoDateTime HTTP/1.1" 200 None
2025/03/08 12:34:55 Status code = 200
2025/03/08 12:34:55 This is what I received: <p>Set Arduino datetime to 20250308123453</p>
2025/03/08 12:34:55 New Arduino time is 20250308123453

greiginsydney, 2025-03-17:
This is a strange one!

The problem with the Pi is that its clock is volatile, so each time it powers up there's no guarantee what the time will be. Inside the Pi this is "fudged" (guessed): it stops the clock when it's turned off, saves that value, and then re-starts from that point the next time it boots.

If the time being saved to the Arduino isn't correct, it might be that the setTime script is running BEFORE the Pi has queried the NTP server for its real time, or maybe that query failed - which could be why this is occurring intermittently.

I'd check the logs to see if the Pi's NTP service is showing any errors:

```
journalctl -u systemd-timesyncd
```

Here's an example of what success looks like in my system:

```
-- Boot d6e077a8f00b4df99f7648a6c62200f8 --
Mar 18 08:18:16 BenchPi3BPlus systemd[1]: Starting systemd-timesyncd.service - Network Time Synchronization...
Mar 18 08:18:17 BenchPi3BPlus systemd[1]: Started systemd-timesyncd.service - Network Time Synchronization.
Mar 18 08:19:02 BenchPi3BPlus systemd-timesyncd[387]: Contacted time server 13.55.50.68:123 (2.debian.pool.ntp.org).
Mar 18 08:19:02 BenchPi3BPlus systemd-timesyncd[387]: Initial clock synchronization to Tue 2025-03-18 08:19:02.718072 A>
pi@BenchPi3BPlus:~ $
```

<br>

\- Greig.

greiginsydney, 2025-07-13:
closing stale issue.

closed 2025-07-13 by greiginsydney